FooDeas/raspberrypi-ua-netinst #101: Install breaks, if package is not available

sidey79, 2017-10-20:
I've got the following error, during Installation of debian stretch on my pi1:

`
Configuring apt:
  Configuring Raspbian repository... OK
  Checking Raspbian repository entry... OK
  Adding raspberrypi.org GPG key to apt-key... OK
  Configuring RaspberryPi repository... OK
  Configuring RaspberryPi preference... OK
  Copying './fhemrepo.list' to /etc/apt/sources.list.d/... OK
  Adding key './fhemrepo.key' to apt...
    Warning: apt-key output should not be parsed (stdout is not a terminal)
    OK
OK
  Copying './99translation' to /etc/apt/apt.conf.d/... OK

Updating package lists... OK

Downloading packages...
  Reading package lists...
  Building dependency tree...
  Reading state information...
  E: Unable to locate package sendEmail
ERROR: 100, trying again (2/5)...   Reading package lists...
  Building dependency tree...
  Reading state information...
  E: Unable to locate package sendEmail
ERROR: 100, trying again (3/5)...   Reading package lists...
  Building dependency tree...
  Reading state information...
  E: Unable to locate package sendEmail
ERROR: 100, FAILED !

Error: The installation could not be completed!
`
I used the same configuration as already done with jessie, only changed the release to stretch.
I figured out now, that the package is renamed to lowercase sendemail.

I open this issue, because i do not feel it correctly, that the installer breaks if a package can't be found. I think it is better, to log that error but got ahead with the installation and try the other packages.

thijstriemstra, 2017-10-25:
I'd like it to break when a package is missing, so perhaps this feature should become a config option.

FooDeas, 2017-10-26:
Ack. The only question is: What should be the default behaviour?

mpmc, 2017-10-26:
@FooDeas As it is now. Break on error.

Mausy5043, 2017-10-26:
I'd prefer it if the installer would continue as long as possible and not give up so soon.

mkayg, 2017-10-29:
I prefer the "fail fast" method that breaks on error

FooDeas, 2017-11-04:
3 vs. 2 so far. Because installations would have to be checked for missing packages by the user - I acknowledge.
Var: `packages_errorignore`
Default: `0`

Coming soon!

closed 2023-08-15 by FooDeas