FooDeas/raspberrypi-ua-netinst #102: Unsigned packages?

dinosore, 2017-10-27:
Doing a stretch build on v2.0.0 on a model B1. I get

```
Configuring apt:
  Configuring Raspbian repository... OK
  Checking Raspbian repository entry... OK
  Adding raspberrypi.org GPG key to apt-key... OK
  Configuring RaspberryPi repository... OK
  Configuring RaspberryPi preference... OK
  Copying './syncthing.list' to /etc/apt/sources.list.d/... OK
  Copying './raspberrypi.gpg' to /etc/apt/trusted.gpg.d/... OK
  Copying './syncthing.gpg' to /etc/apt/trusted.gpg.d/... OK

Updating package lists... OK

Downloading packages...
  Reading package lists...
  Building dependency tree...
  Reading state information...
  The following additional packages will be installed:
    device-tree-compiler libcap2-bin libpam-cap libraspberrypi0
  The following NEW packages will be installed:
    device-tree-compiler libcap2-bin libpam-cap libraspberrypi-bin
    libraspberrypi0 raspberrypi-bootloader raspberrypi-kernel
    raspberrypi-sys-mods raspi-copies-and-fills
  0 upgraded, 9 newly installed, 0 to remove and 4 not upgraded.
  Need to get 36.8 MB of archives.
  After this operation, 132 MB of additional disk space will be used.
  WARNING: The following packages cannot be authenticated!
    raspberrypi-sys-mods raspi-copies-and-fills device-tree-compiler
    raspberrypi-bootloader libraspberrypi0 libraspberrypi-bin raspberrypi-kernel
  E: There were unauthenticated packages and -y was used without --allow-unauthenticated
ERROR: 100, trying again (2/5)...   Reading package lists..
```.
  The copying raspberry.gpg line is my attempt at a quick fix by putting the key in the config.

I'll try Jessie.

FooDeas, 2017-10-27:
Looks like they do some changes or updates or problems in their repo. Not the first time...
If they updated the  stretch packages only, jessie would work. But I propose just to wait some time and try it again later.

dinosore, 2017-10-27:
"some time" ... it's been like this since yesterday at least...

Search on rpi forumdoesn't work for me "high traffic"

And jessie's just failed with the same thing...

[I may be some time ...](https://en.wikipedia.org/wiki/Lawrence_Oates#Last_words)

FooDeas, 2017-10-27:
Hmm... You have a raspberrypi.gpg in your config/apt folder. This could be the problem, too...

FooDeas, 2017-10-27:
Just rename it and try again!

dinosore, 2017-10-27:
Gave that a go.  No luck. Thanks for support.

Trying rpi forums - I may have to wait...

dinosore, 2017-10-27:
Further to a post on rpi forums https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=196336&p=1227771#p1227764 the packages may be ok

FooDeas, 2017-10-28:
To eliminate problems with the config. I'd suggest to delete the contents of the fat partition and extract the installer from zip. Do not make any changes or edit the installer-config and give that a try.
Packages or repo key haven't changed the last week. So it should work as normal. I'll check that on the same hardware today, too.

FooDeas, 2017-10-29:
No problems here.

closed 2017-10-29 by FooDeas

dinosore, 2017-10-29:
Works with base config. I'll dig a bit further.

dinosore, 2017-10-29:
I think I've tracked down the issue, but not the solution (apart from not using the failing feature).

Renaming the config/apt/syncthing.list and config/apt/syncthing.gpg files and commenting out packages=... caused it to work.

I haven't tested leaving the other packages in the config as yet, but don't think that would make a difference.

I think that https://askubuntu.com/questions/31286/additional-keyrings-for-apt-in-trusted-gpg-d may explain the symptoms.

If I understand it correctly, dding a key rather than a keyring to  /etc/apt/trusted.gpg.d can cause key verification to fail on other repositories.  This does seem to match the symptoms I am getting with rhe raspberry pi repository..

A quick glance at the code  didn't find any apt-key add for syncthing.

dinosore, 2017-10-29:
How can it be invalid when it discloses a bug?

FooDeas, 2017-10-29:
https://github.com/FooDeas/raspberrypi-ua-netinst/issues/103

FooDeas, 2017-10-29:
I totally overlooked the problem... I read my code again and I think, I found the "bug"...
From `custom_sources.txt`

> To add a key:
>   Place a file with the extension ".key" in this folder containing the key.
>   It will be imported by the installer.

So all you have to do is to give the keyfile the correct extension ".key".

dinosore, 2017-10-29:
So I confused key and gpg.  At least we both know the symptoms if someone else does the same thing.

I read the paragraph below "To add a gpg:"  Perhaps I wouldn't have made the mistake if it said "To add a keyring:"

I reran with the .key file. and all was ok.

Thanks for the help

dinosore, 2017-10-29:
An aside.

I used
```
hostname=pi

boot_volume_label="${hostname}boot"
rootfs_mkfs_options="-L ${hostname}root"
```
Maybe a root_volume_label might be a little more convenient.

Not important though.

Mausy5043, 2017-10-29:
The `boot_volume_label` will show up in OSX Finder or Windows Explorer when you plug the SD-card in a Mac or PC.
A `root_volume_label` would have less added value I think.

FooDeas, 2017-10-29:
Good ideas! Please create issues for doc improvement and volume label suggestions!

dinosore, 2017-10-30:
> @Mausy5043 Here's how I use partition labels in Raspbian
```
root@pi:~# mount -l|grep '\['
/dev/mmcblk0p2 on / type ext4 (rw,noatime,errors=remount-ro,data=ordered) [piroot]
/dev/sda1 on /mnt/t type fuseblk (rw,relatime,user_id=0,group_id=0,allow_other,blksize=4096) [tdrive]
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro) [piboot]
/dev/sda2 on /mnt/u type ext4 (rw,relatime,data=ordered) [udrive]
root@pi:~# lsblk -f
NAME        FSTYPE LABEL  UUID                                 MOUNTPOINT
sda                                                            
├─sda1      ntfs   tdrive 5A22DB2D22DB0CBF                     /mnt/t
└─sda2      ext4   udrive 9e363742-ab1a-4126-b863-25ca4553dd5d /mnt/u
mmcblk0                                                        
├─mmcblk0p1 vfat   piboot 0C54-CFC0                            /boot
└─mmcblk0p2 ext4   piroot 8e5def0b-e7c6-4017-b7b8-2ce7e1bf7236 /
root@pi:~# cat /etc/fstab
LABEL=piroot		/	ext4		errors=remount-ro,noatime	0	1
LABEL=piboot		/boot   vfat            defaults                        0       0
LABEL=tdrive		/mnt/t  ntfs            nofail                          0       0
LABEL=udrive		/mnt/u	ext4		nofail				0	2

tmpfs			/tmp	tmpfs		defaults,nodev,nosuid		0	0

```
Same goes for linux on pc.