FooDeas/raspberrypi-ua-netinst #111: Clarify wording in config/apt/custom_sources.txt

dinosore, 2017-11-05:
```
# Choose only 1 of the following:
#   To add a key:
#     Place a file with the extension ".key" in this folder containing the key.
#     It will be imported by the installer.
#   To add a **keyring:**
#     Place a file with the extension ".gpg" in this folder containing the keyring.
#     It will be placed in "/etc/apt/trusted.gpg.d/" by the installer.

```
This makes it a bit clearer to me.

It might be nice to add a 3rd option to download the key in the installer.  For example, https://apt.syncthing.net/ gives the instruction to add its release key as
`curl -s https://syncthing.net/release-key.txt | sudo apt-key add -`
Perhaps there could be a file with a list of urls for the installer to download the keys.

Mausy5043, 2017-11-05:
could you not use the `post-install.txt` for that?

dinosore, 2017-11-05:
@Mausy5043 , you can say that for any change. ;-)

Mausy5043, 2017-11-05:
Yes you are right. But, IMHO, whether to add functionality to the installer or require the user to add it in the `post-install.txt` depends on how useful the function will be for *all* users. Every function added means as an increase in size of the installer. It **is** after all 

> The minimal Raspbian unattended netinstaller for Raspberry Pi.

(my) emphasis on "minimal".

dinosore, 2017-11-05:
My view is that it is simpler for the user to provide the address of the key than to have to download and copy it.

To male things simpler and more compact, the other 2 methods could be removed.

closed 2017-12-27 by FooDeas