FooDeas/raspberrypi-ua-netinst #112: Allow installer progress to be monitored from the local betwork

dinosore, 2017-11-05:
I haven't worked out the details, but it might be possible to implement a lightweight solution using nmap.

nmap, I believe, was originally BSD, but has ben widely adopted in Linux.  I don't know about mac.

Windows 10 pcs are covered with embedded Ubuntu.

mpmc, 2017-11-05:
How about http://mqtt.org/ ?

FooDeas, 2018-01-05:
I have an idea implementing this by serving a simple, one directional, buffered telnet connection output.
Coming very soon!

FooDeas, 2018-01-07:
Done with 4c9588e30ab6898bd3353aff3a71a203e42a8e7d.

closed 2018-01-07 by FooDeas

dinosore, 2018-01-21:
Works well.  I'm very impressed with the implementation of this feature.  Well done!

Mausy5043, 2018-12-31:
Can't figure out how this would work on my Mac. Any suggestions?