FooDeas/raspberrypi-ua-netinst #114: Does not work on 256 MB Raspberry Pi B

tdobes, 2017-11-12:
On a Raspberry Pi with 256 MB of RAM, I was not able to get raspberrypi-ua-netinst 2.0.0 to work properly.  I did not use any custom installer-config.txt.

The initial "Copying boot files..." step in install.sh (line 660) fails with an out-of-space error ("No space left on device") due to memory exhaustion.  I was able to work around this by adding "gpu_mem_256=16" to the [all] section of config.txt, allocating more RAM to the CPU (and, therefore, the RAM filesystem).

However, even after doing that, the install process ends up failing on line 1434 of install.sh (cdebootstrap-static).  This failure generally occurs at or slightly after the "P: Configuring helper cdebootstrap-helper-apt" log message with the error "Aborted".  I'm guessing it's running out of RAM.

I have attached the log files both before and after my config.txt tweak:
[error_before_configtxt_tweak.log](https://github.com/FooDeas/raspberrypi-ua-netinst/files/1464269/error_before_configtxt_tweak.log)
[error_after_configtxt_tweak.log](https://github.com/FooDeas/raspberrypi-ua-netinst/files/1464270/error_after_configtxt_tweak.log)

FooDeas, 2017-11-12:
It looks like the sdcard could be the problem. Try with another one.

tdobes, 2017-11-12:
@FooDeas Thank you for the suggestion.  However, I am reasonably certain that the SD card is not the problem in this case.  During my work yesterday, I tested with multiple SD cards, including a microSD card which I verified worked fine with raspberrypi-ua-netinst on a Raspberry Pi 3.  The other cards I used were known to work reliably in these Raspberry Pis as well.  (They previously had a old netinst of wheezy.)  After the errors occurred, I inspected dmesg and did not see any errors indicating issues with the SD cards.  I also observed that the same SD cards worked properly on the older Raspberry Pi's by placing a Raspbian Lite image on them, then removing and installing a number of packages.

FooDeas, 2017-12-10:
Tested on same hardware (256MB RAM) using v2.0.0 without any problems using default configuration.

FooDeas, 2017-12-10:
I'll improve the memory usage soon. But I don't think, it's memory related.

FooDeas, 2017-12-29:
Problem confirmed.

FooDeas, 2017-12-29:
Adding a temporary file based swap on SD card would be a solution if the memory usage improvements in v2.0.1 RC1 are not sufficient.

FooDeas, 2017-12-29:
Fixed with 8db0a1be5ae1559ee8f70553f6961fde4a14b7a4.

closed 2017-12-29 by FooDeas