FooDeas/raspberrypi-ua-netinst #115: Install package from external repository

eLvErDe, 2017-12-08:
Hello,

I tried to create a post-install.sh script but running chroot /rootfs apt-get update fails with apt-key complaining it cannot create /dev/null (wtf ?).

Can you try this ? It's quite important to me as I need to install "puppet" from an external repository so the installation is complete and unattended.

Thanks a lot,

Adam.

FooDeas, 2017-12-09:
`chroot /rootfs /usr/bin/apt-get update` works fine.
Could you upload the post-install.txt file somewhere?

FooDeas, 2017-12-09:
Info: A good method to debug the `post-install.txt` is to remove it completely, set `console` as final action and try the commands step by step.

eLvErDe, 2017-12-09:
Ok thanks. Il see. I tried the config/apt/ folder to setup my repo but it locked during apt update. I'm not sure it's related to your installer, I'll give more feedback soon.

Thanks for the hint

FooDeas, 2017-12-09:
No success yet.

eLvErDe, 2017-12-15:
Hi,

I can confirm installer stalls everytime I add my compagny repository. Commenting the line makes it work, uncommenting after installation: works like a charm.
There's something really weird going on there and I must admit I'm running out of ideas. apt-get update is currently running under strace with the hope I see some interesting lines before it locks...

Would you give me an email address so I can give you the exact lines that breaks everything ?

Best regards, Adam.

FooDeas, 2017-12-16:
You can use the email address from the author info of the git commits.

FooDeas, 2017-12-27:
@eLvErDe Did you send me anything? Nothing received.

closed 2017-12-31 by FooDeas