FooDeas/raspberrypi-ua-netinst #116: Installation not possible with rtc=mcp7941x

eLvErDe, 2017-12-08:
Hi,

hwclock fails to run. Once installed, I added "dtoverlay=i2c-rtc,mcp7941x" to config.txt and it worked just fine.
I guess there's something missing with installer "boot" environment. Maybe i2c is not enabled, or we're missing some overlays. I'll try to look into a it but your help would be welcome.

Regards, Adam.

eLvErDe, 2017-12-08:
Adding

```
dtparam=i2c_arm=on
```

Did not help.

When getting into the emergency initrd shell, hwclock says: cannot access /dev/misc/rtc
dmesg has no reference to i2c or rtc :/

I don't have much more ideas...

FooDeas, 2017-12-09:
I don't see why, because I have no installer-config.txt from you. :)
Did you use `rtc=mcp7941x` there?

eLvErDe, 2017-12-09:
Yes exactly and I noticed the installer restarts once to add dtolverlay=i2c-rtc,mcp7941x but then hwclock does not work.

I replaced "| fail" by "| true" and the installation went fine and clock was working properly at first boot.

FooDeas, 2017-12-09:
Looks like a missing dependency in the stretch installer environment...

eLvErDe, 2017-12-09:
Is there any way to see if the kernel actually sees the I2C clock ?
I'm still thinking about something kernel/overlays related...

FooDeas, 2017-12-09:
i2cdetect from i2c-tools should help here.

eLvErDe, 2017-12-09:
Yes but it's not available in installer initrd

FooDeas, 2017-12-09:
Right you've to copy that in place manually. I'm checking the problem with an ds3231 on a B+ and report.

FooDeas, 2017-12-09:
Problem confirmed.

FooDeas, 2017-12-09:
Checkout and try the current devel branch. I'm still testing but think, it should be fixed.

eLvErDe, 2017-12-09:
Cannot do at the moment. Friday for sure, but I'll try to have a colleague feedback on Monday.

Thanks !

FooDeas, 2017-12-10:
Successfully tested with DS3231 on 7ffcaf3. Please report.

closed 2017-12-10 by FooDeas

eLvErDe, 2017-12-12:
Hi,

Installation seems to start fine when moving to devel branch so I guess it's fixed.
Test has been done by a colleague but we confirmed it was failing, then switched branch and confirm installation continues so I think it's fine.

Thanks a lot for this great fork.

Adam.