FooDeas/raspberrypi-ua-netinst #117: `gpg` guessing command

Mausy5043, 2017-12-09:
While building I get this:

```
Downloading raspbian.public.key.
raspbian.public.key                                100%[================================================================================================================>]   1.73K  --.-KB/s    in 0s
Checking key file 'raspbian.public.key'... gpg: WARNING: no command supplied.  Trying to guess what you mean ...
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
OK
Importing 'raspbian.public.key' into keyring... OK

Downloading raspberrypi.gpg.key.
raspberrypi.gpg.key                                100%[================================================================================================================>]   1.68K  --.-KB/s    in 0s
Checking key file 'raspberrypi.gpg.key'... gpg: WARNING: no command supplied.  Trying to guess what you mean ...
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
OK
Importing 'raspberrypi.gpg.key' into keyring...OK
```

It all still works, but I was wondering if it is known which command `gpg` is guessing at?

FooDeas, 2017-12-09:
There are two commands:
```
gpg --homedir gnupg --keyid-format long --with-fingerprint --with-colons "raspbian.public.key"
gpg -q --homedir gnupg --import "raspbian.public.key"
```
Using GnuPG 1.4.20 on Ubuntu 16.04.3 LTS, no warning occurs. Please provide the GPG version or the "correct" command. Maybe we could adapt it...

Mausy5043, 2017-12-10:
I was building on a Pi 3 `Linux rbdns 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux` with stock `gpg` installed:

```
$ gpg --version
gpg (GnuPG) 2.1.18
libgcrypt 1.7.6-beta
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Home: /home/pi/.gnupg
Supported algorithms:
Pubkey: RSA, ELG, DSA, ECDH, ECDSA, EDDSA
Cipher: IDEA, 3DES, CAST5, BLOWFISH, AES, AES192, AES256, TWOFISH,
        CAMELLIA128, CAMELLIA192, CAMELLIA256
Hash: SHA1, RIPEMD160, SHA256, SHA384, SHA512, SHA224
Compression: Uncompressed, ZIP, ZLIB, BZIP2
```

FooDeas, 2017-12-10:
Please check the newest devel for these or similar warnings.

Mausy5043, 2017-12-10:
Just started a build with branch `devel`:

```
Branch devel set up to track remote branch devel from origin.
Switched to a new branch 'devel'
HEAD is now at 8b576a1 doc: remove unused Code Climate maintainability badge from readme
```

Result:
```
Setting up gpg...
gpg: keybox '/mnt/backup/rbian/builder/raspberrypi-ua-netinst/packages/gnupg/pubring.kbx' created
gpg: /mnt/backup/rbian/builder/raspberrypi-ua-netinst/packages/gnupg/trustdb.gpg: trustdb created

Downloading raspbian.public.key.
raspbian.public.key                     100%[===============================================================================>]   1.73K  --.-KB/s    in 0s
Checking key file 'raspbian.public.key'... gpg: WARNING: no command supplied.  Trying to guess what you mean ...
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
OK
Importing 'raspbian.public.key' into keyring... OK

Downloading raspberrypi.gpg.key.
raspberrypi.gpg.key                     100%[===============================================================================>]   1.68K  --.-KB/s    in 0.001s
Checking key file 'raspberrypi.gpg.key'... gpg: WARNING: no command supplied.  Trying to guess what you mean ...
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
OK
Importing 'raspberrypi.gpg.key' into keyring...OK

Downloading Release file and its signature...
raspberry_Release                       100%[===============================================================================>]  24.14K  --.-KB/s    in 0.01s
raspberry_Release.gpg                   100%[===============================================================================>]     488  --.-KB/s    in 0s
Verifying Release file... OK

Downloading main package list...
tmp.gz                                  100%[===============================================================================>] 123.86K  --.-KB/s    in 0.04s
Verifying main package list... OK
Setting up gpg...

Downloading raspbian.public.key.
raspbian.public.key.1                   100%[===============================================================================>]   1.73K  --.-KB/s    in 0s
Checking key file 'raspbian.public.key'... gpg: WARNING: no command supplied.  Trying to guess what you mean ...
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
OK
Importing 'raspbian.public.key' into keyring... OK

Downloading raspberrypi.gpg.key.
raspberrypi.gpg.key.1                   100%[===============================================================================>]   1.68K  --.-KB/s    in 0.001s
Checking key file 'raspberrypi.gpg.key'... gpg: WARNING: no command supplied.  Trying to guess what you mean ...
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
OK
Importing 'raspberrypi.gpg.key' into keyring...OK
[...]
```

😕

FooDeas, 2017-12-11:
Fixed now. I had to suppress the warning because older GnuPG versions don't support the new commands.

closed 2017-12-11 by FooDeas