FooDeas/raspberrypi-ua-netinst #119: Installer hangs when updating...?

mike632t, 2017-12-24:
When installing on a Raspberry Pi 2 Model B Rev 1.1 everything seems to start off working as expected but the install seems to hang after about 15-20 minutes while updating the kernel.

There does not seem to be an error log created.

Last output is as follows

Configuring apt:
  Configuring Raspbian repository... OK
  Checking Raspbian repository entry... OK
  Adding raspberrypi.org GPG key to apt-key... OK
  Configuring RaspberryPi repository... OK
  Configuring RaspberryPi preference... OK

Updating package lists... OK

Downloading packages...
  Reading package lists...
  Building dependency tree...
  Reading state information...
  The following additional packages will be installed:
    device-tree-compiler libcap2-bin libpam-cap libraspberrypi0
  The following NEW packages will be installed:
    device-tree-compiler libcap2-bin libpam-cap libraspberrypi-bin
    libraspberrypi0 raspberrypi-bootloader raspberrypi-kernel
    raspberrypi-sys-mods raspi-copies-and-fills
  0 upgraded, 9 newly installed, 0 to remove and 4 not upgraded.
  Need to get 36.8 MB of archives.
  After this operation, 132 MB of additional disk space will be used.
  Get:1 http://archive.raspberrypi.org/debian stretch/main armhf raspberrypi-sys-mods armhf 20171127 [9090 B]
  Get:2 http://archive.raspberrypi.org/debian stretch/main armhf raspi-copies-and-fills armhf 0.6 [7226 B]
  Get:3 http://archive.raspberrypi.org/debian stretch/main armhf device-tree-compiler armhf 1.4.4-1 [349 kB]
  Get:4 http://archive.raspberrypi.org/debian stretch/main armhf raspberrypi-bootloader armhf 1.20171029-1 [3513 kB]
  Get:5 http://archive.raspberrypi.org/debian stretch/main armhf libraspberrypi0 armhf 1.20171029-1 [837 kB]
  Get:6 http://archive.raspberrypi.org/debian stretch/main armhf libraspberrypi-bin armhf 1.20171029-1 [335 kB]
  Get:7 http://archive.raspberrypi.org/debian stretch/main armhf raspberrypi-kernel armhf 1.20171029-1 [31.7 MB]

Mausy5043, 2017-12-24:
I get that too sometimes. Then after a couple of hours everything works great. Could be high traffic. Try again at a later moment.

mike632t, 2017-12-27:
I've tried three different models on a couple of different SD cards but the result is the same every time.

FooDeas, 2017-12-27:
It must have something to do with your connection.
Can you download the kernel [manually](http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20171029-1_armhf.deb)?
Are you behind a proxy/firewall?
Which installer version are you using?

mike632t, 2017-12-27:
I am behind a couple of firewalls but I can down load the lernel maunally using my browser and wget (no proxy set).

Using a Pi Zero W with a USB network adapter (wired) with installer version 2.0.0.

No issues getting all the other packages but I wonder if I am running out of memory or disk space ..

mike632t, 2017-12-28:
Full console output attached (password modified)

[2.0.0_console_log.txt](https://github.com/FooDeas/raspberrypi-ua-netinst/files/1590087/2.0.0_console_log.txt)

FooDeas, 2017-12-28:
It's no out of memory or space problem.
Try to disable IPv6.

mike632t, 2017-12-28:
Disabled IPv6 in the installer-config.txt - No change
I can 'see' the traffic on the firewall as the file downloads, and the green LED on the Pi blinks as it is written to the SD card, but then nothing else happens...

Mausy5043, 2017-12-28:
raspberrypi.org still has issues:

```
$ ping archive.raspberrypi.org
ping: archive.raspberrypi.org: Temporary failure in name resolution
$ ping raspberrypi.org
ping: raspberrypi.org: Temporary failure in name resolution
```

This could be related.
You could try pointing your webbrowser at `http://raspberrypi.org` when this happens to confirm the problem.  
Probably lots of people sitting at home trying to do the same thing, overloading that host.

FooDeas, 2017-12-28:
Tricky... Does it keep flashing or stops?
Another idea: Disconnect the LAN port and try to use the integrated wireless:
```
ifname=wlan0
wlan_ssid=<SSID>
wlan_psk=<WPA-PSK>
```

mike632t, 2017-12-28:
It stops - I'll give the interfrated wireless a go in a minute

mike632t, 2017-12-28:
Managed to get 2.0.1 working - and discovered a base install really is basic!!

Thanks for your help

closed 2017-12-28 by mike632t

FooDeas, 2017-12-29:
What was the reason?

mike632t, 2017-12-29:
Succesfully completed a minimal install - but shutdown, reboot halt and poweroff are 'missing' intentional..?

mike632t, 2017-12-29:
Still don't know why it worked..

Mausy5043, 2017-12-29:
I've observed some peculiarities with `raspberrypi.org` too lately. Switching off DNSSEC on my DNS seemed to fix it.

FooDeas, 2017-12-29:
It would be nice if the raspberrypi.org repository would have some official mirrors. 9 of 10 problems are caused by downtime, load or inconsistencies (sometimes caused by my proxy, too).
I set up an "apt-cacher-ng" instance on another pi to catch most of them.

mike632t, 2017-12-29:
Think the problem I was having was as a result of DNS issues when looking up mirrordirector.raspbian.org - it may have been redirecting to domain my firewall doesn't link, but as I use dnsmasq to redirect them to my loopback address I don't think it is logged anywhere!

Mausy5043, 2017-12-29:
@mike632t : you might want to add `log-queries` and optionally `log-dhcp` to your `dnsmasq` config file.

mike632t, 2017-12-29:
Thanks - I'll look into that shortly

FooDeas, 2017-12-31:
I have the same problem today: Hanging on the kernel download.
And it can't be a DNS issue. I checked that using the 8er DNS server.

mike632t, 2017-12-31:
Glad it isn't just me!!  Unfortunatly I'm no closer to working out what it might be as there was nothing in my dnsmasq logs.

FooDeas, 2017-12-31:
Download freezed at 23MB.
Successfully tried again two hours later.

Mausy5043, 2017-12-31:
I'm almost sure that these problems will be much less after 02JAN2018.

FooDeas, 2017-12-31:
My experience from today:
5 fails in a row. 1 "success" with issues at (kernel) boot. Similar to that what @mike632t reported.

My "analysis":
I disabled my transparent proxy: 5 "real" successes in a row
I re-enabled the transparent proxy: fail at downloading kernel...
I re-enabled my apt-cacher-ng (with proxy): success

mike632t, 2017-12-31:
Which proxy are you using..?

FooDeas, 2017-12-31:
Squid

eLvErDe, 2018-09-19:
@FooDeas again you're live save. I experencied the same insane shit, raspberrypi-kernel stuck downloading when going throught Squid 3.5 (CentOS 7). It worked after changing config to use apt-cacher-ng instead of Squid.

mike632t, 2019-10-22:
Same problem again with buster - but this is the only time I have an issue installing Debian from behind my firewall. A direct conenction to my router solved the problem (again), might be worth mentioning in a readme...?