FooDeas/raspberrypi-ua-netinst #120: Test v2.0.1

FooDeas, 2017-12-27:
Could some of you help testing v2.0.1?
I pushed it to the branch "devel_untested". Have a look in it's changeling, too.

mike632t, 2017-12-27:
Pi Zero W with USB ethernet

Scripts worked fine - created a partition of aprox 133MB.

Seems to be a problem when the install script tried to run though - attached output from console.

[2.0.1_console_log.txt](https://github.com/FooDeas/raspberrypi-ua-netinst/files/1590024/2.0.1_console_log.txt)

FooDeas, 2017-12-27:
@mike632t Thank you! This should be fixed now.

mpmc, 2017-12-28:
Tried on my old rpi b & it fails unfortunately. I captured a video [here](http://markclarkstone.co.uk/static/video.webm) - There's some weird artefacts not related to the install just vlc & udp not playing nice.

Logs below..

[error-20171228T000304.log](https://github.com/FooDeas/raspberrypi-ua-netinst/files/1590094/error-20171228T000304.log)
[error-20171228T000410.log](https://github.com/FooDeas/raspberrypi-ua-netinst/files/1590095/error-20171228T000410.log)
[error-20171228T000516.log](https://github.com/FooDeas/raspberrypi-ua-netinst/files/1590096/error-20171228T000516.log)

FooDeas, 2017-12-28:
I see... Fix for this comes tomorrow.

mpmc, 2017-12-28:
@FooDeas Ok will be more than happy to test.

mike632t, 2017-12-28:
Pi Zero W - looks like I get the same issue
[error-20171228T001626.log](https://github.com/FooDeas/raspberrypi-ua-netinst/files/1590122/error-20171228T001626.log)
[error-20171228T001722.log](https://github.com/FooDeas/raspberrypi-ua-netinst/files/1590123/error-20171228T001722.log)
[error-20171228T001823.log](https://github.com/FooDeas/raspberrypi-ua-netinst/files/1590124/error-20171228T001823.log)

Mausy5043, 2017-12-28:
[L1003](https://github.com/FooDeas/raspberrypi-ua-netinst/blame/devel_untested/scripts/opt/raspberrypi-ua-netinst/install.sh#L1003) should be `-L` instead of `-l`

FooDeas, 2017-12-28:
@Mausy5043 That's not the point. It's much more annoying. :)

Should be fixed now.

Mausy5043, 2017-12-28:
Ah yes. I see.

Mausy5043, 2017-12-28:
That fixed it. Just got a succesfull boot on my Pi
```
Revision 396b52c
Built on Thu Dec 28 10:50:39 CET 2017
Running on Raspberry Pi version B+
```
No errors.

mpmc, 2017-12-28:
Can confirm latest git works on my B 2.0 ([000e](https://elinux.org/RPi_HardwareHistory)) too. Will dig out my 0002 model  & report back.

FooDeas, 2017-12-28:
Thanks!
I'll release this as v2.0.1 RC1.

FooDeas, 2017-12-28:
@mpmc I'm lookup forward to the results with the 0002 model.

FooDeas, 2017-12-28:
@Mausy5043 Could you check, if a custom `raspberrypi-ua-netinst/config/wpa_supplicant.conf` is still working?

Mausy5043, 2017-12-28:
building off `devel` right?

FooDeas, 2017-12-28:
Yes. It's identical to the untested branch.
This version has small code changes but these could have much impact...

mpmc, 2017-12-28:
@FooDeas Not good I'm afraid. [Video here](http://markclarkstone.co.uk/static/0002.mp4).

FooDeas, 2017-12-29:
@mpmc That's weird... I tested this with a much more limited RAM setting. But nevertheless the 256MB version of the model B seems to have these problems.
Could you enter the console and post the output of "free -h" and "du -h" please?
Also have a try with a new line `gpu_mem=16` to the "all"-section of the config.txt.

Mausy5043, 2017-12-29:
## RaspberryPi 3
Tried installing on a Pi3 with my normal WiFi settings:

```
Network configuration:
  ifname = wlan0
  ip_addr = dhcp
  wlan_ssid = 
  wlan_psk = 
  online_config = 
```

SSID and PSK are supplied in `wpa_supplicant.conf` instead of via the setttings in `installer-config.txt`.
This did not work (by design?) as this part of the log shows:

```
Loading additional kernel modules:
  Loading module '8192cu'... OK

Waiting for wlan0... OK
Starting wpa_supplicant... Successfully initialized wpa_supplicant
Line 4: Invalid passphrase length 0 (expected: 8..63) '"'.
Line 4: failed to parse psk '""'.
Line 5: failed to parse network block.
Failed to read or parse configuration '/tmp/wpa_supplicant.conf'.
nl80211 driver didn't work. Trying generic driver (wext)...
Successfully initialized wpa_supplicant
Line 4: Invalid passphrase length 0 (expected: 8..63) '"'.
Line 4: failed to parse psk '""'.
Line 5: failed to parse network block.
Failed to read or parse configuration '/tmp/wpa_supplicant.conf'.

Error: The installation could not be completed!
```

I think it's ignoring the `wpa_supplicant.conf`

I then tried by adding the SSID and PSK to the `installer-config.txt`.
This worked. So, for Pi3 `wpa_supplicant.conf` is not needed/working. WiFi settings should be put in `installer-config.txt`.

## RaspberryPi 1B+ (with WiFi dongle)
Tried installing on a Pi1B+ with a WiFi dongle and my normal WiFi settings:

```
Network configuration:
  ifname = wlan0
  ip_addr = dhcp
  wlan_ssid = 
  wlan_psk = 
  online_config = 
```

SSID and PSK are supplied in `wpa_supplicant.conf` instead of via the setttings in `installer-config.txt`.
This also did not work:

```
modprobe: can't load module rt2800lib (kernel/drivers/net/wireless/ralink/rt2x00/rt2800lib.ko): unknown symbol in module, or unknown parameter
Loading additional kernel modules:
  Loading module '8192cu'... OK

Waiting for wlan0... 1.. 2.. 3.. 4.. 5.. 6.. 7.. 8.. 9.. 10.. 11.. 12.. 13.. 14.. FAILED

Error: The installation could not be completed!
```

Realising that I might have mismatched the WiFi dongle and the kernel module I then swapped out the WiFi dongle for another one.

```
modprobe: can't load module rtl8192cu (kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192cu/rtl8192cu.ko): Device or resource busy
Loading additional kernel modules:
  Loading module '8192cu'... OK

Waiting for wlan0... OK
Starting wpa_supplicant... Successfully initialized wpa_supplicant
Line 4: Invalid passphrase length 0 (expected: 8..63) '"'.
Line 4: failed to parse psk '""'.
Line 5: failed to parse network block.
Failed to read or parse configuration '/tmp/wpa_supplicant.conf'.
nl80211 driver didn't work. Trying generic driver (wext)...
Successfully initialized wpa_supplicant
Line 4: Invalid passphrase length 0 (expected: 8..63) '"'.
Line 4: failed to parse psk '""'.
Line 5: failed to parse network block.
Failed to read or parse configuration '/tmp/wpa_supplicant.conf'.

Error: The installation could not be completed!
```

No joy there either. 
I think (here too) it's ignoring the `wpa_supplicant.conf`. 

So, finally, I then tried by adding the SSID and PSK to the `installer-config.txt`.
This worked. So, also for a RPi with a dongle `wpa_supplicant.conf` is not needed/working. 
WiFi settings should be put in `installer-config.txt`.

As far as I'm concerned I can live with the supplied `wpa_supplicant.conf` not working. 
I'll just have to find a way to inject the SSID and PSK into the `installer-config.txt` when (re)building.
That should not be too hard.

## Conclusion
Installing on a Pi3 using WiFi and on a Pi 1B+ with WiFi dongle (and the correct `drivers_to_load`) works, 
provided that the SSID and PSK are supplied via `wlan_psk` and `wlan_ssid` in `installer-config.txt`

Mausy5043, 2017-12-29:
It seems to me that this:
https://github.com/FooDeas/raspberrypi-ua-netinst/blob/6fc4391cc4aed2517581d4de126810af459e2661/scripts/opt/raspberrypi-ua-netinst/install.sh#L493-L494
together with this:
https://github.com/FooDeas/raspberrypi-ua-netinst/blob/6fc4391cc4aed2517581d4de126810af459e2661/scripts/opt/raspberrypi-ua-netinst/install.sh#L683
means that the path to `wlan_configfile` is invalid.
Therefore, this will always be `true`
https://github.com/FooDeas/raspberrypi-ua-netinst/blob/6fc4391cc4aed2517581d4de126810af459e2661/scripts/opt/raspberrypi-ua-netinst/install.sh#L803
and the custom `wpa_supplicant.conf` will never be found.

Probably L494 should read:  
```
wlan_configfile=/raspberrypi-ua-netinst/config/wpa_supplicant.conf 
```

@FooDeas : what do you think?

FooDeas, 2017-12-29:
@Mausy5043 You're right. I pushed a possible fix (and some more changes) into the "wpaconf"-branch. Could you give it a go?

Mausy5043, 2017-12-29:
@FooDeas : Doesn't work.

https://github.com/FooDeas/raspberrypi-ua-netinst/blob/a3a4ac38e1052bbae4e5e86eec3fbf0fd34d09e4/scripts/opt/raspberrypi-ua-netinst/install.sh#L804-L806

If you could change both `"${tmp_bootfs}"/boot/raspberrypi-ua-netinst` into `"${tmp_bootfs}"/raspberrypi-ua-netinst` that would probably fix it. ;-)

FooDeas, 2017-12-29:
So true... done.

OT: I had some issues with my local git config. Sorry for the force pushes!

Mausy5043, 2017-12-29:
Install started succesfully.

FooDeas, 2017-12-29:
I hope, it will complete.
If so and @mpmc could successfully test the 0002 model against the current devel branch, I will rise the version.

Mausy5043, 2017-12-29:
Found another issue:
"unexpected token '&' in line 1401"

https://github.com/FooDeas/raspberrypi-ua-netinst/blob/7367b7ea6192ff246a123d521b1bb0e5f8d3a68e/scripts/opt/raspberrypi-ua-netinst/install.sh#L1401

space between `2>` and `&1` is not allowed.

FooDeas, 2017-12-29:
Thanks. Fixed with dfe22adbe1ab2ff8518dd53271638cc1c087d23a.

mpmc, 2017-12-29:
> I hope, it will complete.
If so and @mpmc could successfully test the 0002 model against the current devel branch, I will rise the version.

Building & testing now :)

Mausy5043, 2017-12-29:
Successful install on Pi1B+ with WiFi dongle and custom `wpa_supplicant.conf`.
👍

FooDeas, 2017-12-29:
@Mausy5043 Thanks a lot!
@mpmc I prepared a build archive for you: [raspberrypi-ua-netinst-git-75552e0.zip](https://github.com/FooDeas/raspberrypi-ua-netinst/releases/download/v2.0.1_RC1/raspberrypi-ua-netinst-git-75552e0.zip) I hope, this will fix this. I don't have a model with 256MB RAM here.

mpmc, 2017-12-29:
@FooDeas testing now..

FooDeas, 2017-12-29:
Wow. I didn't really think that would work as expected... 😄
Thanks a lot!

mpmc, 2017-12-29:
Here's the full log, http://termbin.com/0qe3

FooDeas, 2017-12-29:
I'm interested in that what comes between "Installation finished at Fri Dec 29 19:13:16 UTC 2017 and took 20 min 2 sec (1202 seconds)" and the reset.
But no rush - this part will be released as it is!

The only new feature that follows will be a overwritable subnet mask calculation.

FooDeas, 2017-12-29:
Release: This year!

mike632t, 2017-12-29:
Think I have a problem with the following 

Revision v2.0.1_RC1 (0f3c408)
Built on Thu Dec 28 13:45:07 CET 2017
Running on Raspberry Pi version Zero W

Minimal install completes succesfully (including some additional packages) using on-board wireless and the Pi Zero W powers off at the end.

However when I reapply power the only thing that happens is that the green LED flashes breifly, after that nothing else happens (no output on serial console no further activity on LED).  If power down and insert the DS card in a Pi version 2 model B it works just fine but will not work in the Pi Zero W.

Will test in a Pi 3 shortly

FooDeas, 2017-12-30:
That happens if the Pi doesn't load the bootloader.bin and/or the start.elf from SD. normally not installer related.
Wait for the v2.1.0 release.

mike632t, 2017-12-30:
Don't know if I get the rainbow screen or not, can only connect using a serial console.  

Very happy to wait for 2.1.0

FooDeas, 2017-12-31:
v2.1.0 released. Thanks a lot to you all!

closed 2017-12-31 by FooDeas

mpmc, 2017-12-31:
@FooDeas Glad to help. Just ping me if you'd like me to test anything on the 0002 :)