FooDeas/raspberrypi-ua-netinst #121: No 'shutdown' command in minimal install

mike632t, 2017-12-29:
Assume deliberatly omitted..?

FooDeas, 2017-12-29:
"ls /sbin/shutdown"?

mike632t, 2017-12-29:
Not there...

[2.0.0_ls.txt](https://github.com/FooDeas/raspberrypi-ua-netinst/files/1592440/2.0.0_ls.txt)

FooDeas, 2017-12-29:
There should be a symlink to /bin/systemctl...
Please attach `/var/log/raspberrypi-ua-netinst.log`.

mike632t, 2017-12-29:
Had to redo the install to produce logs, selecting to delete them after the install was a bad idea, but I now know that whatever the problem is it is reproducable!  

[2.0.0_ls.txt](https://github.com/FooDeas/raspberrypi-ua-netinst/files/1593182/2.0.0_ls.txt)
 [2.0.1-installer-config.txt](https://github.com/FooDeas/raspberrypi-ua-netinst/files/1593176/2.0.1-installer-config.txt)
[2.0.1-raspberrypi-ua-netinst.log ](https://github.com/FooDeas/raspberrypi-ua-netinst/files/1593177/2.0.1-raspberrypi-ua-netinst.log)

FooDeas, 2017-12-29:
Thanks a lot!
The `systemd-sysv` package is part of the server preset only. That was forgotten in the docs because I didn't distinguish between the init systems. You can add it to the packages in the installer-config if you need it.

I'll revise the docs according this.

closed 2017-12-29 by FooDeas

mike632t, 2017-12-30:
Awsome installer - very happy running stretch on my original Pi B.  The install doesn't take too long and stretch it boots suprisingly quickly too.

Many thanks for your efforts.  Happy New year!

SlySven, 2018-11-27:
Only recently discovered this project but as I am a staunch SysV `init` die-hard who avoids letting the `systemd` hydra anywhere near the PID 1 spot - I just wanted to advise that I recently managed to force a `systemd` less install with a:
```
preset="minimal"
packages="sysvinit-core,sysvinit-utils,sysv-rc,vim-tiny,iputils-ping,wget,ca-certificates,rsyslog,cron,dialog,locales,tzdata,less,man-db,logrotate,bash-completion,console-setup,apt-utils,libraspberrypi-bin,raspi-copies-and-fills,vim,aptitude"
```
I.e. use do everything else from the `preset="server"` case but avoiding supping from *that* bottle of cool-aide.

I've run into problems with the no shutdown, no reboot situation on, I think it was a Linux laptop - IIRC there was a `poweroff` command that did work though.  I suspect it was probably induced in a Debian Wheezy to Stretch `dist-upgrade` which got messed up part way through or got confused because by default that version change (from {Deb|Raspb}ian 8.x to 9.x) tries to switch you from systemV `init` to `systemd`  unless you are really, *really, **firm*** about not letting it...