FooDeas/raspberrypi-ua-netinst #122: P3 / Pi Zero W don't boot after completing install of 2.1.0

mike632t, 2017-12-31:
Install completes successfully on both, but when the reboot the green LED just flashes breifly and nothing else happens.  (I suspect that some firmware / kernel modules may be missing)

Both installations boot normally on a Pi 2 B.

FooDeas, 2017-12-31:
I'm afraid that this is not version related. I had that yesterday before I solved my connection problems. Confirmed with the last two versions, too.
I disabled my transparent proxy and things were working as expected.

mike632t, 2017-12-31:
As I suspected - is it fixable (using the Pi 2 B?).  I'm assuming that one or more packages didn;t download

FooDeas, 2017-12-31:
logfile?

FooDeas, 2017-12-31:
Or you can try to setup apt-cache-ng.

FooDeas, 2017-12-31:
Interesting info:

> AIUI the raspberry pi foundation repo serves small files directly but under some circumstances redirects large files to a third party CDN.

[Source](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=83491&sid=4dba589a74e405c009a90a54d8c0c626&start=50#p1153085)

This explains why the problems only occur on two packages.

FooDeas, 2017-12-31:
I'm testing the following solution using a mirror I finally found:

Create the file "/raspberrypi-ua-netinst/config/apt/raspberrypi.org.list". Content:
`deb http://mirrors.ustc.edu.cn/archive.raspberrypi.org/debian __RELEASE__ main`

Result in 20 minutes.

mike632t, 2018-01-02:
Did some more testing normally I enable the uart as I only use the serial console - when I do this the LED on my Pi zero flashes breifly and it doesn't boot - if I don't enable the UART the LED flashes beifly then stays on...  I suspect it is booting (or starting to) but I can;t find it on my network!

FooDeas, 2018-01-04:
Could you attach your `installer-config.txt`?

mike632t, 2018-01-04:
Added logs and config fiels to post

FooDeas, 2018-01-04:
Try this:

**config.txt**
Use installer **default** file without any changes.

**cmdline.txt**
`dwc_otg.lpm_enable=0 consoleblank=0 console=serial0,115200 elevator=deadline rootwait` (nothing else)

**installer-config**
Use yours but **remove** the `cmdline` setting.

mike632t, 2018-01-04:
Don't I neet to use enable_uart=1 for the serial console on the Pi Zero to work?

mike632t, 2018-01-04:
As I was beginning to suspect the problem is something to do with the enable_uart setting.

I now have a HDMI - VGA adapter so I was able to repeat the install on a Pi3, but this time I modified config.txt and cmdline.txt so that when it booted it used the standard console.

Initially with enable_uart=1 the Pi hung after ~0.3 seconds (kernel panic).  Powered down, changed config.txt so enable_uart=0 and tried again - success!!

So the question is what is missing to enable it to use the serial console...?

FooDeas, 2018-01-04:
Related:
https://github.com/raspberrypi/firmware/issues/798

FooDeas, 2018-01-04:
Just try the settings I posted above and report if the console is still working.

mike632t, 2018-01-05:
Following the advise [in the related issue](https://github.com/raspberrypi/firmware/issues/798) I removed 'kgdboc=serial0,115200' from cmdline.txt and it works on both my Pi3 and Pi Zero W.

I Still need to sort it out so I get the kernel messages on the console but basically I'm a very happy bunny!  I'm not sure how long it takes a Pi 3 to boot with a minimal install - but it isn't long!

closed 2018-01-05 by mike632t

FooDeas, 2018-01-05:
Related: https://github.com/raspberrypi/linux/issues/2245