FooDeas/raspberrypi-ua-netinst #123: No /boot/raspberrypi-ua-netinst/reinstall/config.txt

kraeml, 2018-01-04:
I installed ua-netinst several times on PI3 but with the new version I couldn't found the file:

`/boot/raspberrypi-ua-netinst/reinstall/config.txt`

There is no dir `/boot/raspberrypi-ua-netinst/`

I this file gone? How to I reinstall ua-netinst 2.1.0?

FooDeas, 2018-01-04:
I'm pretty sure, I missed that. Take the one from the installer zip.

FooDeas, 2018-01-05:
Did you use the cleanup=1 setting in your config?

mike632t, 2018-01-05:
cleanup=1
cleanup_logfiles=1

FooDeas, 2018-01-05:
The cleanup option removes **all** installer related files from the boot partition.
For a reinstallation you have to copy them to the SD card again.

kraeml, 2018-01-05:
Yes, indeed. I will change this in my script into `cleanup_logfiles`.

Thanks for your help.

closed 2018-01-05 by kraeml