FooDeas/raspberrypi-ua-netinst #124: Logging to syslog-ng server

Mausy5043, 2018-01-08:
`rsyslog` is installed by default when installing `preset=server`.  
I'd like to set-up a `syslog-ng` server elsewhere on my LAN and have it suck up all the logging from the other clients for centralised review.  
Any suggestions how I could easily install `syslog-ng` instead of `rsyslog` as the log-sender?

On a related side-note: It would be interesting if the the installer logs could be sent to a `syslog-ng` server. 😉

FooDeas, 2018-01-08:
Without any installer modifications, couldn't you set the minimal preset and add the missing server packages with the packages variable - replacing rsyslog with syslog-ng and providing the config file(s)?

To implement a syslog sender for the installer logs looks a bit like overkill at first glance. I thought about a very tiny solution and decided to implement a telnet log output last weekend.
Have you tried that already? I know, there are some disadvantages comparing to a client solution but I hope this is a useful feature, client OS independent and user friendly (as far as you know the IP address).

Mausy5043, 2018-01-09:
> Without any installer modifications, couldn't you set the minimal preset and add the missing server packages with the packages variable - replacing rsyslog with syslog-ng and providing the config file(s)?

👍 Now why hadn't I thought of that? 😄 

Re the telnet log. I actually normally only check the logs once the system is installed. Or, if something goes wrong I need to take the Pi to my desk where I can connect it to an old 15" VGA screen. Most of my Pies are in hard to reach places, so I've never taken the UART-route. The telnet option I have not yet tried out. I'll give it a try next time.

closed 2018-01-09 by Mausy5043