FooDeas/raspberrypi-ua-netinst #126: read-only option

aristosv, 2018-01-17:
For the past 3 years I’ve been running 45 raspberry pi 2B’s at various locations throughout my country. They are mostly used for playing music and videos in retail shops. I noticed that depending on the location’s conditions and the fact that the operating system is running off a microSD, I would get frequent corruptions of the OS. I’ve (mostly) resolved the problem by running the OS as read-only.

To do this I’ve used raspbian-ua-netinst to create a minimal baseline image, and redirected any process trying to write on the microSD to /tmp, which was mounted on tmpfs. Then modified /etc/fstab to set / and /boot as read-only. This really reduced the corruption issues, since nothing was written on the microSD.

Now I’m looking to upgrade the pi’s to 3B, and I want to rebuild my baseline image to the latest raspbian version. I noticed that raspbian-ua-netinst is not really under active development, since it only supports the Rpi 2B. So I stumbled upon this project and I wanted to ask:

Is it possible to add an option to build an image that will run as read-only?

FooDeas, 2018-01-19:
You can add eventually needed further packages, supply mounting scripts and update the initramfs from the a post-install script.

If you provide your working installer-configuration, I would like to add this as a new feature.

aristosv, 2018-01-19:
This is pretty much how I would handle it in post-install.txt
I'm not sure it's the correct way, but so far it's working.

```
echo -n "Configuring ReadOnly..."
  rm -rf /rootfs/var/log/
  ln -s /tmp /rootfs/var/log
  
  rm -rf /rootfs/var/run/
  ln -s /tmp /rootfs/var/run
  
  rm -rf /rootfs/var/lock/
  ln -s /tmp /rootfs/var/lock
  
  rm -rf /rootfs/var/lib/dhcp/
  ln -s /tmp /rootfs/var/lib/dhcp
  
  rm -f /rootfs/etc/issue
  ln -s /tmp/issue /rootfs/etc/issue
  
  rm -f /rootfs/etc/fake-hwclock.data
  ln -s /tmp/fake-hwclock.data /rootfs/etc/fake-hwclock.data
  
  sed -i 's/vfat defaults/vfat defaults,ro/g' /rootfs/etc/fstab
  sed -i 's/errors=remount-ro/defaults,ro/g' /rootfs/etc/fstab
echo "OK"

echo -n "Configuring DNS..."
  rm /rootfs/etc/resolv.conf
  ln -s /tmp/resolv.conf /rootfs/etc/resolv.conf
  sed -i 's/\/etc\/resolv.conf/\/tmp\/resolv.conf/g' /rootfs/sbin/dhclient-script
echo "OK"
```

Mausy5043, 2018-01-19:
You should not mount those on `/tmp` but instead mount those as `tmpfs`.

Add something like this in `/etc/fstab`:

```
tmpfs   /var/log    tmpfs    defaults,nodev,nosuid    0   0
```
Options may need some tweaking.

jacobalberty, 2018-01-25:
In my experience a pure read only root is difficult to get working. Things like to break left and right. I couldn't get systemd-timesyncd working using a similar setup to the listed one.

I went the overlayfs route and wasn't too hard to get working with raspberrypi-ua-netinst. I just needed to add initramfs-tools to the packages and put this script in /etc/initramfs-tools/scripts/init-bottom/root-ro . add the right config.txt entries
```
initramfs initrd followkernel
ramfsfile=initrd
ramfsaddr=-1
```
add the right cmdline.txt parameter `root-ro-driver=overlay` and then the following in post-install.txt
```
# May not need all of these (or even any) mounted for mkinitramfs
mount -o bind /dev /rootfs/dev
mount -o bind /sys /rootfs/sys
mount -o bind /proc /rootfs/proc

# Include the overlay module in initrd
echo overlay >> /rootfs/etc/initramfs-tools/modules
# generate the initrd
chroot /rootfs /usr/sbin/mkinitramfs -o /boot/initrd

umount /rootfs/dev
umount /rootfs/sys
umount /rootfs/proc
```

This makes everything think it has write access to the system so everything "just works" without needing any services to be modified.

aristosv, 2018-03-01:
any plans on implementing this?

positron96, 2020-03-14:
Hi, I just wanted to say to those who consider RO root unreliable - I had my RPi working for a couple of years with RO / and /boot and RW /var and /home. It was working fine until my /var partition broke down. So, nothing breaks if whole /var is RW.
To make a fully RO root I'd advise to mount /var as tmpfs on top of SD card /var via overlay fs. Or something like that. I have no idea how to make it in ua-netinst configs though.

closed 2021-01-04 by FooDeas