FooDeas/raspberrypi-ua-netinst #127: Installer fails on mkfs.vfat (devel branch)

dinosore, 2018-01-21:
Using a 16GB sd card formatted as a single FAT32 partition, mkfs.vfat failed.

```
Waiting for /dev/mmcblk0... 
  Disk /dev/mmcblk0: 14.4 GiB, 15502147584 bytes, 30277632 sectors
  Disklabel type: dos
  Disk identifier: 0x4e42d3df
OK
Applying new partition table... OK
Initializing /boot as vfat... mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows
mkfs.vfat: Too many clusters for filesystem - try more sectors per cluster
mkfs.fat 4.1 (2017-01-24)

Error: The installation could not be completed!
```
The log extract above is with the redirection to /dev/null removed.

Removing -s 1 fixed the problem.  I can't see why this should be - I suspect an mkfs bug.

Pull request follows.

zejn, 2018-01-22:
One sector per cluster fixes an issue with RPi3 compute modules, see https://github.com/FooDeas/raspberrypi-ua-netinst/pull/125

Can you try adding "-v" to mkfs.vfat and then showing the output?

FooDeas, 2018-01-22:
@dinosore Could you check it again against devel?

dinosore, 2018-01-22:
Some sd card problems at the moment.  Will try again as soon as I can.

dinosore, 2018-01-23:
I've updated my fork as per [](https://github.com/FooDeas/raspberrypi-ua-netinst/commit/b6cd12dcd12040d47c530fde502fd4b42fc51521) 

Haven't been able to test yet - hope to do so later today.

dinosore, 2018-01-24:
I've tested and it works!

It might be better, though, not to suppress mkvs messages.  Your call.

closed 2018-01-24 by dinosore

FooDeas, 2018-01-24:
Thanks a lot! I'll merge it soon.

PR #128 (2018-02-18): Devel
1. Move assignment statements out of loop
2. Fix problem with mkfs.vfat
3 Readme clarification.

fixes #127