FooDeas/raspberrypi-ua-netinst #131: Issues with 2.1.0

thijstriemstra, 2018-01-22:
I tried devel but things failed so I tried 2.1.0 but seeing similar errors there:

```

==================================================
raspberrypi-ua-netinst
==================================================
Revision v2.1.0 (d50d417)
Built on ma 22 jan 2018 21:48:04 CET
Running on Raspberry Pi version 3 Model B
==================================================
https://github.com/FooDeas/raspberrypi-ua-netinst/
==================================================
Starting HWRNG... OK
Mounting boot partition... OK
Copying boot files... OK
Executing installer-config.txt...
dos2unix: can't open '/bootfs/raspberrypi-ua-netinst/config/installer-config-common.txt': No such file or directory
tail: can't open '/bootfs/raspberrypi-ua-netinst/config/installer-config-common.txt': No such file or directory
tail: no files
/tmp/bootfs/raspberrypi-ua-netinst/config/installer-config.txt: line 2: /bootfs/raspberrypi-ua-netinst/config/installer-config-common.txt: No such file or directory
```

Second issue:

```
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 4.9.59+ /boot/kernel.img
  run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 4.9.59-v7+ /boot/kernel7.img
  Processing triggers for man-db (2.7.6.1-2) ...
  Setting up device-tree-compiler (1.4.4-1) ...
  Setting up libraspberrypi0 (1.20171029-1) ...
  Setting up raspberrypi-sys-mods (20180103) ...
  Fixing file capabilities...
  Failed to set capabilities on file `/bin/ping6' (Invalid argument)
  The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file
  Failed to set capabilities on file `/bin/ping' (Operation not supported)
  The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file
  Failed to set capabilities on file `/usr/bin/systemd-detect-virt' (Operation not supported)
  The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file
```

3rd issue:

```
  Adjusting cmdline.txt...
  Removing old sysctl tweaks, if present...
  Removing old NOPASSWD line, if present...
  sed: can't read /etc/sudoers: No such file or directory
  Failed
```

Any ideas @FooDeas?

thijstriemstra, 2018-01-22:
Looks like issue nr 1 should be (when looking at [this diff](https://github.com/FooDeas/raspberrypi-ua-netinst/compare/v2.0.0...master#diff-df2fc3ca35b9e979ba4b7cabd2108dc9R346) be fixed by replacing `/boot/` with `/rootfs/boot/`. Testing..

FooDeas, 2018-01-25:
The logged errors from the second and third issue are caused by the kernel post install script. They came with the latest updates and can be ignored.

thijstriemstra, 2018-01-25:
Did anything change when it comes to swap/memory because my rpi3 always freezes during installation of packages.. tried different rpi's, different sd-cards, nothing helps. Will try with 2.0.0 to see if that fixes it.

FooDeas, 2018-01-25:
I tested my rpi3 without problems. Swap is not used on models with enough memory.

thijstriemstra, 2018-01-25:
Hmm, here's my config:

```
preset=server
release=stretch
firmware_packages=1
packages=git,unzip,nano,libffi-dev,libi2c-dev,i2c-tools,libbz2-dev,pciutils,libsqlite3-dev,libncurses-dev,usbutils,libusb-dev,libyaml-dev,libts-dev,xz-utils,raspberrypi-sys-mods,bluez,bluetooth,pi-bluetooth,libbluetooth3,libbluetooth-dev,libdbus-c++-dev,libdbus-1-dev,libinput-dev,libinput10,libwebp-dev,lsyncd,samba,autofs,cifs-utils,libspeechd-dev,flite,flite-dev,libreadline-dev,libraspberrypi-dev,raspberrypi-kernel-headers,libssl-dev
usersysgroups=lpadmin,lp,dialout,input,audio,mail,spi,i2c
hostname=HOST_NAME
userperms_admin=1
root_ssh_pwlogin=0
ssh_pwlogin=0
user_ssh_pubkey=mykey_rsa.pub
timezone=Europe/Amsterdam
locales="en_US.UTF-8,nl_NL.UTF-8"
system_default_locale="en_US.UTF-8"
gpu_mem=512
spi_enable=1
i2c_enable=1
i2c_baudrate=400000
sound_enable=1
sound_usb_enable=1
sound_usb_first=1
userperms_sound=1
camera_enable=1
camera_disable_led=1
usergpio=1
usergpu=1
mirror_cache=CACHE_IP:3142
console_blank=0
quiet_boot=1
disable_raspberries=1
disable_splash=1
cleanup=1
cleanup_logfiles=0
```

I will try again with an empty package list, see if that works.

thijstriemstra, 2018-01-26:
I limited the custom packages to:

```
packages=git,unzip,nano
```

And this time the installer didn't hang... ugh. I'll now try installing the remaining packages from that list (in one command) and see if that hangs the RPI and if it's related to a particular package.

closed 2018-01-26 by thijstriemstra

FooDeas, 2018-01-26:
That's not the reason. The upgrades come from the added repository to apt. The debootstrap step has no access to them. That's another improvement todo...