FooDeas/raspberrypi-ua-netinst #133: Composite video

thijstriemstra, 2018-01-28:
I have this screen that requires the following settings in `/boot/config.txt`:

Disable `hdmi_force_hotplug` and enable `hdmi_ignore_hotplug`:

```
  # enable if hdmi display is not detected and composite is being output
  #hdmi_force_hotplug=1

  # uncomment to force composite output
  hdmi_ignore_hotplug=1
```
And set the screen with and height::
```
  # uncomment to force a console size. By default it will be display's size minus
  # overscan.
  framebuffer_width=320
  framebuffer_height=240
```
And enable overscan::
```
  # uncomment this if your display has a black border of unused pixels visible
  # and your display can output without overscan
  disable_overscan=1
```

What settings would I use here? Note that there is a `hdmi_disable_overscan` option in raspberrypi-ua-netinst but `disable_overscan` is not hdmi-only and also used for composite.

thijstriemstra, 2018-01-29:
I also need `display_rotate` to rotate or flip the screen orientation. The default value is 0, see https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

FooDeas, 2018-04-28:
I'm had a closer look on this today and I'm working on it!
Thanks for your PR! I'm starting with that and push a test version for you very soon.

closed 2023-08-15 by FooDeas