FooDeas/raspberrypi-ua-netinst #135: No reboot, no shutdown commands after install.

Cryosim, 2018-02-05:
reboot
-bash: reboot: command not found
shutdown -r now
-bash: shutdown: command not found
sudo /sbin/shutdown -r now
sudo: /sbin/shutdown: command not found
sudo /sbin/reboot
sudo: /sbin/reboot: command not found

FooDeas, 2018-02-05:
release=buster?
Fix it and try again please.

Cryosim, 2018-02-05:
Why? Buster avaliable. http://mirrordirector.raspbian.org/raspbian/dists/buster/

FooDeas, 2018-02-05:
Buster is not released yet.
Unfortunately not supported.

Cryosim, 2018-02-05:
OK. Can I install release Sid?

FooDeas, 2018-02-05:
No. Same reason. But you can upgrade your stretch installation afterwards.

Cryosim, 2018-02-05:
I install stretch. same issue.
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Reading package lists... Done
reboot
-bash: reboot: command not found

shutdown -r
-bash: shutdown: command not found

/sbin/reboot
-bash: /sbin/reboot: No such file or directory

/sbin/shutdown
-bash: /sbin/shutdown: No such file or directory

sudo reboot
sudo: reboot: command not found

Cryosim, 2018-02-05:
If I install base system, cannot connect via ssh

FooDeas, 2018-02-05:
As you can see in the docs, there is no openssh-server in the base preset.

FooDeas, 2018-02-05:
See #121, too.

closed 2018-02-05 by FooDeas