FooDeas/raspberrypi-ua-netinst #138: f2fs can't be loaded

sidey79, 2018-03-08:
https://github.com/FooDeas/raspberrypi-ua-netinst/blob/d50d417d22d8fd4ba47af24fca153f517d6dda80/scripts/opt/raspberrypi-ua-netinst/install.sh#L1503

Script hangs on release 2.1.0 while trying to load f2fs module.

Bevore it can't change to the right verson of the libs.
4.9.59 is on the filesystem but it tries for 4.9.27

FooDeas, 2018-03-16:
I'll check that next week. Did you build the installer by yourself or used the latest release?

sidey79, 2018-03-16:
I used the latest Release .

sidey79, 2018-03-18:
I have currently no other available.
But with Release 2.0 IT works

Mausy5043, 2018-03-18:
Remarkable. I've built from the `devel` branch this weekend several times without any issues.

napo67, 2018-03-20:
I saw this issue when I was doing a reinstall, so it definitely exists. The issue is also reproducible every time.

sidey79, 2018-03-20:
Yes, that ist exactly what i tried. Reinstall a cortupted System on a new SD card

napo67, 2018-03-20:
I was doing a reinstall as described here since I was updating my installer-config.txt.

https://github.com/FooDeas/raspberrypi-ua-netinst#reinstalling-or-replacing-an-existing-system

kpfleming, 2018-03-27:
I just did two installs from the `devel` branch using f2fs and they worked fine.

napo67, 2018-03-27:
If you build from the devel it will include 4.9.80-v7+ modules so you will not see this issue. It will only happen with the pre-built images.

napo67, 2018-03-30:
Just tried to reinstall from devel branch. It worked once, did not work the second time around:

depmod: can't change directory to 'lib/modules/4.14.30-v7+': No such file or directory
modprobe: can't change directory to '4.14.30-v7+': No such file or directory

kpfleming, 2018-03-30:
The RPF kernel was just updated from 4.9.x to 4.14.x, but I was able to build and use an installer today with the new kernel, so I don't think that's the problem.

napo67, 2018-03-31:
I think it is. If one does not do a new build when the kernel changes, it will cause issues with reinstall. These updates were available for my rpi3, but I did not do the update because of reinstall:

        libraspberrypi-bin 1.20180328-1
        libraspberrypi0 1.20180328-1
        raspberrypi-bootloader 1.20180328-1
        raspberrypi-kernel 1.20180328-1
        raspberrypi-sys-mods 20180328+1

After the reinstall, the second reinstall attempt failed, probably because the ua-netinst upgraded the packages. I guess the reinstall is using the upgraded kernel and that causes the failure.

kpfleming, 2018-03-31:
It appears that something has changed recently and now the `cpio` package must be explicitly installed, otherwise the build.sh script produces a broken installer. Please confirm that you have that package installed.

napo67, 2018-03-31:
The `cpio` package is not installed on my setup. I will check with the release version with `cpio` installed to see if that makes a difference.  Thank you for the info.

Mausy5043, 2018-03-31:
The `cpio` package has not been part of a default installation for a long time.  
When I developed my script to build the installer on a RPi automagically I always made sure to first install `git`, `curl`, `zip` and `cpio`. Usually the first three are already present when I start a build.

FooDeas, 2018-04-03:
I'll prepare a RC of the latest devel. I think that will fix this problem.

thijstriemstra, 2018-04-03:
Or is this a duplicate of #147?

FooDeas, 2018-04-04:
Please test with the pre-release: https://github.com/FooDeas/raspberrypi-ua-netinst/releases/tag/v2.2.0_RC1

napo67, 2018-04-08:
The pre-release installer hangs when it is installing libjavascriptcoregtk-4.0-18.

closed 2018-04-25 by FooDeas