FooDeas/raspberrypi-ua-netinst #144: Missing 'gettext' and 'iw' packages

kpfleming, 2018-03-27:
When installing either the minimal or server preset for any Debian release later than wheezy, the installer adds `raspberrypi-sys-mods' to the postinstall package list (this is not configurable). This package includes a bunch of things, including the `wifi_country.sh` script which is added to /etc/profile.d. Unfortunately this script requires both `gettext` and `iw`, but the package does not have those dependencies stated, and neither of these packages are included in any of the presets.

As a result, on each login, or sudo session, the user sees a bunch of messages about missing tools. If these tools are installed, but the wifi_country has not been set, the user sees a message about it not being set. Some of us don't use WiFi on our RPis, so all of this is just annoyance, but at least these two packages should be installed if raspberrypi-sys-mods is installed.

thijstriemstra, 2018-03-27:
Can confirm this. Installing `libreadline-dev` fixed it.

kpfleming, 2018-03-27:
I just installed 'gettext' and 'iw' and it resolved the problem, although I then blocked raspberrypi-sys-mods because I don't want any of the things it provides, and I don't use wifi so I don't need the wifi_country bits.

FooDeas, 2018-04-03:
I would comment out the script call during the install action.
Agreements?

thijstriemstra, 2018-04-03:
@FooDeas sounds good.

FooDeas, 2018-04-04:
Done with 6d349226134eb7c1617fbb86e6d0fae020bfed32.

closed 2018-04-04 by FooDeas