FooDeas/raspberrypi-ua-netinst #147: Missing `cpio`

Mausy5043, 2018-03-31:
Possibly related to #144:

After updating this morning using `apt` I'm now on kernel xxx and find that I can nolonger succesfully complete a build on my Pies.

Apparently `cpio` is nolonger available:
```
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package cpio is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
   tar

E: Package 'cpio' has no installation candidate
```

`cpio` is required to [complete the build](https://github.com/FooDeas/raspberrypi-ua-netinst/search?utf8=%E2%9C%93&q=cpio&type=).

Mausy5043, 2018-03-31:
Well blow me down! 😮 

`sudo apt update` fixed this.
I was convinced I'd done that.

kpfleming, 2018-03-31:
I just ran into this as well, I did not have the `cpio` package installed. Apparently it's no longer part of the default installation in Stretch.

closed 2018-04-04 by FooDeas