FooDeas/raspberrypi-ua-netinst #149: /etc/resolv.conf should be copied into rootfs for all network configurations

kpfleming, 2018-04-03:
https://github.com/FooDeas/raspberrypi-ua-netinst/blob/09fa962e3b6c461e5c7235d54173c3fbe1ad5920/scripts/opt/raspberrypi-ua-netinst/install.sh#L1800

I recently tried a DHCP-configured install using this tool for the first time (all the rest were using static addressing), and the installation of raspberrypi-kernel and raspberrypi-bootloader packages failed because there was no `/etc/resolv.conf` in the rootfs. As a result, the `chroot apt-get` commands were unable to find package repositories.

I'd send a PR for this, but it seems like an intentional choice, so I opened this issue instead to ask if anyone knows why this logic exists; it's possible that some other part of the script is expected to put a resolv.conf into rootfs when DHCP is used and that's not happening.

FooDeas, 2018-04-03:
Normally udhcpc creates the file if it receives a nameserver.

kpfleming, 2018-04-03:
Yep, and this happens in the in-memory filesystem where the installer is operating. It will also happen in the installed root filesystem when the machine is rebooted after the installation, but during the installation the root filesystem won't have any resolver information automatically configured.

In raspbian-ua-netinst the installer script *always* copies the information into the rootfs to be used until the system is rebooted. I have only been using this installer for a couple of weeks so I can't guess what may have changed here, though.

FooDeas, 2018-04-03:
I had a look at the raspbian-ua-netinst. Theoretically, it should have the same issue...
I don't see why this makes problems now... Just let's move the cp including the dhcp-if before the repository update.

thijstriemstra, 2018-04-03:
Had exact same issue. Thanks for figuring this out @kpfleming

closed 2018-04-04 by FooDeas