FooDeas/raspberrypi-ua-netinst #152: reinstall not working

TimVNL, 2018-04-25:
Revision v2.1.0 (d50d417)
Built on Sun Dec 31 12:06:43 CET 2017
Running on Raspberry Pi version 3 Model B

Starting HWRNG... OK
Mounting boot partition... OK
Copying boot files... OK
Executing installer-config.txt...
OK

Searching for deprecated variables...

Checking if config.txt needs to be modified before starting installation...
  Setting HDMI options... OK
OK

Unmounting boot partition... OK

Network configuration:
  ifname = eth0
  ip_addr = dhcp
  online_config = 

depmod: can't change directory to 'lib/modules/4.14.34-v7+': No such file or directory
modprobe: can't change directory to '4.14.34-v7+': No such file or directory
Waiting for eth0... OK
Configuring eth0 with DHCP... 192.168.178.6
Set time using ntpdate... OK

Installation started at Wed Apr 25 15:30:19 UTC 2018 (UTC).

Resulting installer configuration:
  preset = minimal
  packages = dirmngr,nano,systemd-sysv,iputils-ping,wget,ca-certificates,rsyslog,cron,dialog,locales,tzdata,less,logrotate,bash-completion,console-setup,apt-utils,libraspberrypi-bin,raspi-copies-and-fills
  firmware_packages = 0
  mirror = http://mirrordirector.raspbian.org/raspbian/
  mirror_cache = 
  release = stretch
  hostname = timv-web01
  domainname = 
  user_ssh_pubkey = 
  root_ssh_pubkey = 
  root_ssh_pwlogin = 1
  ssh_pwlogin = 
  username = 
  usergpio = 
  usergpu = 
  usergroups = 
  usersysgroups = 
  userperms_admin = 0
  userperms_sound = 0
  cdebootstrap_cmdline = --flavour=minimal --include=locales,kmod,libpam-systemd,rng-tools,cpufrequtils,ifupdown,net-tools,openssh-server,dosfstools,fake-hwclock,isc-dhcp-client
  packages_postinstall = dirmngr,nano,systemd-sysv,iputils-ping,wget,ca-certificates,rsyslog,cron,dialog,locales,tzdata,less,logrotate,bash-completion,console-setup,apt-utils,libraspberrypi-bin,raspi-copies-and-fills,raspberrypi-bootloader,raspberrypi-kernel,raspberrypi-sys-mods
  boot_volume_label = TIMV-WEB01
  root_volume_label = 
  bootsize = +128M
  bootoffset = 8192
  rootsize = 
  timeserver = time.nist.gov
  timezone = Etc/UTC
  rtc = 
  dt_overlays = 
  keyboard_layout = 
  locales = en_US.UTF-8
  system_default_locale = en_US.UTF-8
  wlan_country = 
  ip_ipv6 = 1
  cmdline = dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 elevator=deadline fsck.repair=yes
  drivers_to_load = 
  gpu_mem = 16
  console_blank = 0
  hdmi_type = 
  hdmi_tv_res = 1080p
  hdmi_monitor_res = 1024x768
  hdmi_disable_overscan = 0
  hdmi_system_only = 0
  usbroot = 0
  usbboot = 0
  rootdev = /dev/mmcblk0
  rootpartition = /dev/mmcblk0p2
  rootfstype = f2fs
  rootfs_mkfs_options = 
  rootfs_install_mount_options = noatime
  rootfs_mount_options = noatime
  final_action = reboot
  quiet_boot = 1
  disable_raspberries = 1
  disable_splash = 1
  cleanup = 0
  cleanup_logfiles = 0
  spi_enable = 0
  i2c_enable = 0
  i2c_baudrate = 
  sound_enable = 0
  sound_usb_enable = 0
  sound_usb_first = 0
  camera_enable = 0
  camera_disable_led = 0

OTP dump:
  16:00280000
  17:1020000a
  18:1020000a
  27:00002727
  28:d8879269
  29:27786d96
  30:00a02082

Waiting 5 seconds.....
Waiting for /dev/mmcblk0... 
  Disk /dev/mmcblk0: 14.9 GiB, 15931539456 bytes, 31116288 sectors
  Disklabel type: dos
  Disk identifier: 0xb860177f
OK
Applying new partition table... OK
Initializing /boot as vfat... OK
Copying /boot files in... OK
Loading f2fs module... 
Error: The installation could not be completed!

thijstriemstra, 2018-04-25:
@TimVNL not sure what's going on but you can also checkout the `devel` branch, build from scratch and try that..

FooDeas, 2018-04-25:
Try the RC!

FooDeas, 2018-04-25:
Duplicate of #138

FooDeas, 2018-04-27:
This is caused by package changes in the debian repository and fixes in devel and latest RC.

closed 2018-04-27 by FooDeas