FooDeas/raspberrypi-ua-netinst #155: Reinstall boot loop on B+

edlins, 2018-05-24:
Hello!

I am trying raspberrypi-ua-netinst for the first time today and it booted up fine on the first try on my B+.  However, when I `cp /boot/raspberrypi-ua-nettinst/reinstall/config.txt /boot` and `reboot`, it goes into a boot loop and gives up after the max retries.  There are three `error*.log` files in `raspberrypi-ua-netinst` that appear to be identical, and one is attached below.  Please let me know what you recommend for performing reinstalls.

[error-20180524T193203.log](https://github.com/FooDeas/raspberrypi-ua-netinst/files/2036868/error-20180524T193203.log)

edlins, 2018-05-24:
This appears to be a dup of #138 and #152.  I'll try v2.2.0 RC1 as suggested and update.

edlins, 2018-05-24:
Also fails in v2.2.0 RC1.

[error-20180524T204857.log](https://github.com/FooDeas/raspberrypi-ua-netinst/files/2037047/error-20180524T204857.log)

edlins, 2018-05-26:
Thank you for your attention to this.  It may not seem like a big deal, but it is to me.  I'm developing an extensive post-install.txt and being able to reinstall remotely through SSH saves me a lot of time over re-imaging the SD card on my laptop.  And right now my laptop SD reader is in use so I can't reinstall at all.  LMK if there's anything I can do to help.

thijstriemstra, 2018-05-31:
B+? can you be more specific?

edlins, 2018-05-31:
https://www.raspberrypi.org/products/raspberry-pi-1-model-b-plus/
```
root@pi:~# cat /proc/version 
Linux version 4.14.34+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1110 Mon Apr 16 14:51:42 BST 2018
root@pi:~# cat /proc/cpuinfo 
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 0010
Serial		: 00000000416b91f0
```

edlins, 2018-06-21:
FYI, poked around at this today.  The failure is from `modprobe` complaining that it can't find folder `4.14.34+` when the highest version modules folder is `4.14.30+`.  It appears that the reinstall is using the installed kernel version (34) but the installer only has modules for the installer version (30).

FooDeas, 2018-06-23:
Workaround: replace the content of the boot partition with the latest release using your config
Reinstall feature is not working atm.

edlins, 2018-06-25:
> Workaround: replace the content of the boot partition with the latest release using your config

Yup, I've been doing that but I have to pull the SD and reflash every time I want to test a change in post-install.txt.  Related, I tested on just-released 2.2 and it made it through a re-install (eventually).  My guess is that this will start failing when a new kernel package comes out (post .34).  I also got a bunch of these messages on the telnet interface (love that!):
```
ERROR: 100, trying again (5/5)...   Reading package lists...
  Building dependency tree...
  Reading state information...
  Package raspberrypi-bootloader is not available, but is referred to by another package.
  This may mean that the package is missing, has been obsoleted, or
  is only available from another source
  Package raspberrypi-kernel is not available, but is referred to by another package.
  This may mean that the package is missing, has been obsoleted, or
  is only available from another source
  E: Package 'raspberrypi-bootloader' has no installation candidate
  E: Package 'raspberrypi-kernel' has no installation candidate
  E: Unable to locate package raspberrypi-sys-mods
  E: Unable to locate package raspi-copies-and-fills
ERROR: 100, FAILED !

Error: The installation could not be completed!
  2 retries left.
  You have 10 seconds to press ENTER to get a shell or it will be retried automatically.
Connection closed by foreign host.
```
Is there something screwy with the repos today?

FooDeas, 2018-11-14:
Not reproducible with v2.2.1.

FooDeas, 2018-11-16:
The initramfs has to match the kernel. That's the problem.
That's a big problem...

FooDeas, 2018-11-16:
It should work now in many cases. This is how to try:
Build the current devel and replace the "raspberrypi-ua-netinst/raspberrypi-ua-netinst.cpio.gz" on the SD card with the new file from the zip file.
Please test and report!

closed 2018-12-05 by FooDeas

edlins, 2019-02-12:
Confirmed fixed in v2.3.0 release, thanks!!  For my own understanding, it looks like you did two things here:
1) Added code to `install.sh` that checks for a mismatch bw kernel in uname (arm v6 or v7) and the modules folder (also v6 or v7) and if mismatched makes a symlink to match the uname.  Can a v6 kernel really run the v7 modules and vice-versa?
2) Put both modules folders (v6 and v7) in the image.  Seems safer but consumes an additional 50+MB in the image?  Since this happens before the network is up they have to already be there, right?
```
root@pi:/lib/modules# du -s *
54643	4.14.79+
56404	4.14.79-v7+
```
Please excuse my ignorance, but I'd like to better understand this useful install process.

FooDeas, 2019-02-12:
The modules are loaded from the cpio but the kernel could have been updated by the system. That's the problem.
If they mismatch, I create a symlink that lets the newer kernel use the old modules. In most cases, that should work. But of course a v6 kernel will trigger only a v6 module symlink and a v7 kernel only a v7 symlink. Notice the "v" in the grep command of the else block.
The safest solution would be to backup complete installer environment including the kernel, device tree files, etc. But yes, that would consume drive space.

I think, the current solution is a reasonable compromise that works in many cases.