FooDeas/raspberrypi-ua-netinst #156: Use hostname provided by DHCP

mrichar1, 2018-05-30:
Hi,

I was wondering if it might be possible to have the option to use a hostname provided by DHCP if one is provided, rather than having to hard-code it?

I've had a look at `udhcpc` and it seems to set the variable `$hostname` if one is provided by DHCP, which can then be used via the `default.script` it calls. If there was some way to flag to the installer to use this variable, that would be great!

Happy to dig into the code and see how feasible this is if you think it's a good idea?

kpfleming, 2018-05-30:
Certainly seems like a reasonable thing to do, behind an opt-in configuration variable.

mrichar1, 2018-05-30:
So having dug a bit more there are actually quite a few variables which are offered by DHCP which might be useful to use to dynamically override those in `installer-config.txt`, including (but possibly not limited to):

* hostname
* domain
* ntp/time server

( https://github.com/jgunthorpe/busybox/blob/master/networking/udhcp/README.udhcpc)

I'm now wondering if instead of a specific setting for each one, it might be better to have an option that takes a list of variables to source from DHCP?  Something like:

`dhcp_options=hostname,domain,timeserver`

And then have a mapping of dhcp options to netinst options to override.