FooDeas/raspberrypi-ua-netinst #159: rt2870 firmware

edlins, 2018-06-03:
Hi there, I'm trying to perform a net install over Wi-Fi using a rt2870 USB device but the interface doesn't come up in the installer with the message:
```
modprobe: can't load module rt2800lib (kernel/drivers/net/wireless/ralink/rt2x00/rt2800lib.ko): unknown symbol in module, or unknown parameter
```
Once the fresh install is booted (via ethernet) I can bring up the interface but I have to first load the ralink firmware with:
```
sudo apt-get install firmware-misc-nonfree
```
Can you confirm that this firmware is not included?  If so is there any chance that a nonfree installer will be provided, or do I get to develop and maintain that?  These devices are cheap and plentiful so I imagine lots of users could benefit from this.

Thanks!

kpfleming, 2018-06-03:
That package is definitely not included in the standard installer image.

edlins, 2018-06-03:
Okay, thanks.  It's a catch-22 that you need the network to install the network so maybe this note will help others avoid it.  Don't expect a network install with the standard installer using a rt2870!!

closed 2018-06-03 by edlins