FooDeas/raspberrypi-ua-netinst #161: WiFi firmware for RPi 3 Model B+ not included in release 2.2.0

DaStoned, 2018-07-25:
Booting release 2.2.0 on RPi 3B+ using WiFi fails with:

`Waiting for wlan0... 1.. 2.. 3.. 4.. 5.. 6.. 7.. 8.. 9.. 10.. 11.. 12.. 13.. 14.. FAILED`

The kernel doesn't seem to find the WiFi drivers (probably the firmware). I see the following kernel message immediately before the raspberrypi-ua-netinst  comes up:

```
brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43455-sdio.bin for chip 0x004345(17221) rev 0x000006
brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.bin failed with error -2
```

I switched to booting from Ethernet and - while boot succeeded - the same message was in logs and the Wifi interface was nowhere to be seen. Only after installing the raspbian package "firmware-brcm80211" and manually configuring wpa_supplicant did the Wifi interface appear and connect to my network.

installer-config for failed boot
```
hostname=ts99

packages=nano,htop,usbutils,libnss-mdns,avahi-utils,apt-utils,psmisc,gcc,wpasupplicant,wireless-tools,raspi-config

dt_overlays="pi3-disable-bt"

ifname=wlan0
wlan_ssid="my_ssid"
wlan_psk="my_passwd"
wlan_country="EE"
```

A side note - I would really, really like to boot and install my RPi-s en masse (that's why I'm using this fine automated installer, after all) on my desk using Ethernet and subsequently move them to their final location where they'll be using WiFi. This WiFi is not available where I'm doing the installation work. Is that not possible ATM?

DaStoned, 2018-07-26:
Thank you for the quick response - I will be able to test the RC next Monday.

DaStoned, 2018-07-30:
Fix confirmed, thank you. Using RC2 the WiFi on RPi 3B+ works out of the box.
I disconnected Ethernet, ran the installer. Both installer and post-boot normal mode see the WiFi interface and connect to the network I have here (using the same config as above).

closed 2018-07-30 by DaStoned

FooDeas, 2018-07-30:
Thanks for the feedback (and your email)! I'll merge and release it today evening.