FooDeas/raspberrypi-ua-netinst #162: Feature: configure a WiFi network without mandating its use during installation

DaStoned, 2018-08-08:
Hi!

A development idea: I'd like to describe a wireless network in the user friendly "installer-config.txt" file for future use. Then I'd like to be able to pop the SD card into the RPi and install raspbian while connected to Ethernet. This small detail is important, as I might not have the described WiFi network available to me while installing the system. Only after finishing the installation would I move the RPi to the location where the WiFi network is available.

Bonus trick: I could describe multiple WiFi networks instead of a single one.

Please feel free to consider this a low urgency feature, as a simple workaround is available: I can set up and copy my own "/etc/wpa_supplicant/wpa_supplicant.conf" and "/etc/network/interfaces.d/50-wlan0" files with desired content, while stripping all wlan0 stuff out of "installer-config.txt":

```
country=EE

network={
	scan_ssid=1
	ssid="MyWifiSSID"
	psk="MyWifiPass"
	key_mgmt=WPA-PSK
}
```
and
```
allow-hotplug wlan0
iface wlan0 inet dhcp
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```

Mausy5043, 2018-08-08:
IMHO you have described the designed behaviour.
You either set-up the wireless network and install with it OR you supply the required files using the "Bring Your Own Files" mechanism. The latter is not meant to be a workaround. 😉

DaStoned, 2018-08-08:
Oh, perhaps. It wasn't obvious to me, though.

iiidefix, 2019-05-12:
If one wants to use WiFi after installing from a Ethernet only config following additional packages are required:
```
firmware-brcm80211 wpasupplicant crda
```

closed 2023-08-15 by FooDeas