FooDeas/raspberrypi-ua-netinst #163: Fails at "Copying boot files" on RPi version 1B

markslater, 2018-09-01:
I'm unable to get versions > 1.5.2 to work on my RPi 1B.

My understanding of what volumes are mounted at boot time is a bit shaky, but it appears to be a problem with insufficient space on `/tmp/bootfs`

I'm flashing the SD card like so:
```bash
wget -qO- https://github.com/FooDeas/raspberrypi-ua-netinst/releases/download/v2.2.1/raspberrypi-ua-netinst-v2.2.1.img.xz | xzcat - > /dev/mmcblk0
```

When I boot the RPi from the SD card, it stops with SOS quickly.  The error log on the SD card is as follows:

```
==================================================
raspberrypi-ua-netinst
==================================================
Revision v2.2.1 (1c98620)
Built on Mon Jul 30 20:22:39 CEST 2018
Running on Raspberry Pi version B
==================================================
https://github.com/FooDeas/raspberrypi-ua-netinst/
==================================================
Starting HWRNG... OK
Mounting boot partition... OK
Copying boot files... cp: error writing '/tmp/bootfs/start_x.elf': No space left on device

Error: The installation could not be completed!```

For reference, here's what `cpuinfo` gives me on the Pi:
```bash
root@pi:~# cat /proc/cpuinfo
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 0002
```
Note that on version 1.5.2, the install succeeds, but only if I restrict the size of additional files I include - a 2 KB text file works fine, but a 5 MB jar causes it to fail in much the same way as above.  I tried altering the `bootsize` - both increasing (`bootsize=+256M`) and decreasing (`bootsize=+64M`), but no dice.

As I mentioned I don't really understand what volumes are available at that point in the script (I added a `df -h` to the script, but that fails... I assume this is because of where it is in the boot process), nor how to alter the size of the volume.

FooDeas, 2018-09-02:
The installer runs out of memory before SD card initialization.
Do you use any special config?

markslater, 2018-09-03:
No, just the vanilla 2.2.1 image flashed to the SD card.

I've just tried 2.1.0, and that seems to work fine.  However, vanilla 2.0.0 fails with an almost identical log to 2.2.1:

```bash
==================================================
raspberrypi-ua-netinst
==================================================
Revision v2.0.0 (84cbb40)
Built on Mon Oct 16 02:04:49 CEST 2017
Running on Raspberry Pi version B
==================================================
https://github.com/FooDeas/raspberrypi-ua-netinst/
==================================================
Starting HWRNG... OK
Mounting boot partition... OK
Copying boot files... cp: error writing '/bootfs/raspberrypi-ua-netinst/raspberrypi-ua-netinst.cpio.gz': No space left on device
cp: error writing '/bootfs/raspberrypi-ua-netinst/error-19700101T000010.log': No space left on device
cp: error writing '/bootfs/start.elf': No space left on device
cp: error writing '/bootfs/start_cd.elf': No space left on device
cp: error writing '/bootfs/start_db.elf': No space left on device
cp: error writing '/bootfs/start_x.elf': No space left on device

Error: The installation could not be completed!
```

Mausy5043, 2018-09-04:
Could be a dud SD-card

markslater, 2018-09-10:
Sorry for the delay in replying.

I don't think it's a dud SD card - if I put the card in a Pi Zero, it completes fine:
```bash
==================================================
raspberrypi-ua-netinst
==================================================
Revision v2.2.1 (1c98620)
Built on Mon Jul 30 20:22:39 CEST 2018
Running on Raspberry Pi version Zero
==================================================
https://github.com/FooDeas/raspberrypi-ua-netinst/
==================================================
[etc]
```
FWIW I'm loving the project, though I'm currently limited 2.1.0.

mpmc, 2018-11-11:
I can confirm this, I dug out my old 1B and I'm now getting the the same error. 
```
==================================================
raspberrypi-ua-netinst
==================================================
Revision v2.2.1 (1c98620)
Built on Mon Jul 30 20:22:39 CEST 2018
Running on Raspberry Pi version B
==================================================
https://github.com/FooDeas/raspberrypi-ua-netinst/
==================================================
Starting HWRNG... OK
Mounting boot partition... OK
Copying boot files... cp: error writing '/tmp/bootfs/start_x.elf': No space left on device

Error: The installation could not be completed!
```

Checked SD card for errors, fine. Also ran in a zero, that ran fine!

FooDeas, 2018-11-14:
The old 256MB RAM model B gets out of memory...

FooDeas, 2018-11-14:
I had to fake the low memory on my old 1B with 512MB. It should be fixed in devel branch now.
Please test and report!

markslater, 2018-11-14:
Works like a charm.  Thanks!

FooDeas, 2018-11-15:
Thank you! I'll release it.

closed 2018-11-15 by FooDeas