FooDeas/raspberrypi-ua-netinst #165: ufw:

oupala, 2018-11-19:
I have installed raspberrypi-ua-netinst and I'm testing to configure iptables using ufw.

I get an error while configuring it:

```log
 modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/4.14.71+/modules.dep.bin'
modprobe: FATAL: Module ip6_tables not found in directory /lib/modules/4.14.71+\nip6tables v1.6.0: can't initialize ip6tables table `filter': Table does not exist (do you need to insmod?)
Perhaps ip6tables or your kernel needs to be upgraded.
```

Are you aware of any problem about iptables or ufw or ip6tables or the default kernel?

FooDeas, 2018-11-19:
I've never seen that before.
What installer version was used? What's the content of /lib/modules/?

oupala, 2018-11-20:
I think I used the default install of raspberrypi-ua-netinst v2.2.1 (How can I check the exact version I used for install ?).

And here is the content of `/lib/modules/`:

```bash
>ls -l /lib/modules/
total 7
drwxr-xr-x 3 root root 3488 nov.  17 07:06 4.14.79+
drwxr-xr-x 3 root root 3488 nov.  17 07:06 4.14.79-v7+
```

Mausy5043, 2018-11-20:
Let me just leave this here: [hint link](https://unix.stackexchange.com/questions/308129/iptables-could-not-open-moddep-file)

FooDeas, 2018-11-20:
Thanks @Mausy5043! Of course, rebuilding the modules database is an option. But it shouldn't be necessary.
I tried to reproduce the problem. But everything's working as intended and the ip6_tables module loads as expected.
Could you @oupala attach the installer log file from "/var/log/raspberrypi-ua-netinst.log"? Just remove eventually secret data from it before.

closed 2018-12-05 by FooDeas

oupala, 2019-03-22:
Sorry for answering so late. I'm now giving you the log file you asked for:

[raspberrypi-ua-netinst.log](https://github.com/FooDeas/raspberrypi-ua-netinst/files/2996743/raspberrypi-ua-netinst.log)

Is it possible to re-open the issue?

As my install was a standard one, I don't think it is a problem specific to me.

I can proceed to tests if needed.

FooDeas, 2019-03-22:
Could you test it again with the current release?

FooDeas, 2019-03-22:
Don't forget to install iptables via apt or installer configuration. iptables is not installed by default.

oupala, 2019-03-30:
I tester again with a fresh install of raspberrypi-ua-netinst v2.3.0 and everything worked like a charm. Can you please tell me what have changed?

And I did not need to install iptables explicitly because I'm using it through ufw. So I'm installing ufw which depends on iptables.