FooDeas/raspberrypi-ua-netinst #168: Running A Script On First Reboot

backXslash, 2019-03-11:
I'm working on a project where I need to run a script to finish system configuration **after** the first time the system reboots. 

Typically, this would be accomplished by either adding a line to `/etc/rc.local` or by adding an `@reboot` line to the root crontab. 

IIRC, `rc.local` is no longer used on Stretch, it's been replaced by `systemd`. The root crontab is still an option though *(I think)* by using the "Bring Your Own Files" option and putting the pre-configured crontab file in `/var/spool/cron/crontabs/root`. 

I've tried this last one, and it doesn't *appear* to work. Is there a better option that someone can point me to?

tl;dr I just need to run a script the first time the system boots *after* the unattended installation has completed.

kpfleming, 2019-03-11:
You could drop a systemd unit file into `/etc/systemd/system` with `ConditionFirstBoot=true`. See this page: https://www.freedesktop.org/software/systemd/man/systemd.unit.html

backXslash, 2019-03-11:
@kpfleming That will work! Thank you so much!

closed 2019-03-11 by backXslash