FooDeas/raspberrypi-ua-netinst #169: rcS custom file not executing after boot

pculler, 2019-03-13:
I have created an rcS file and as it says on the documentation I placed the file on /raspberrypi-ua-netinst/rcS which is the same installer as in the source code but with some few minor changes just to suit my personal needs. According to the installation log, the file is never take into consideration although it does exists. I have a few questions about it though which are:

- Does the rcS file works when there is an installer-config.txt that I have already created.
- Does it have to do that the file is exactly the same as the main installer with only a few changes?

Thanks!

FooDeas, 2019-05-26:
Please check the installer screen. If the installer detects the file, it executes it **instead** almost the whole normal installation routine. Check the install.sh and search for "rcS" source to get an idea.

FooDeas, 2019-05-26:
You could add a sleep command at the beginning to check, if it really is not recognized and executed.

closed 2019-06-25 by FooDeas