FooDeas/raspberrypi-ua-netinst #170: Pi Zero*

mocallins, 2019-05-25:
Am I to understand, this installer, should work with a Raspberry Pi Zero * ?

I've downloaded the installer.
Followed all instructions, but the Pi does not boot up, as I have seen it do with another image (Raspbian Stretch, on an 8GB card)
I am trying to use a 4GB card (of which I have a bagful) and none of them work ( I don';t blioeve that ALLL are defective), as well as the other 5 Pi's I have to work with.

I don't see any l;ogfile created on the sd card.

I am at a loss.

FooDeas, 2019-05-26:
It definitely works on all Pi Zero models. Maybe you extracted into a subfolder or used a non FAT partition.

mocallins, 2019-05-26:
Nope, root folder.Tried formatting with all FAT types, no luck.Edited installer-config.txt to specifically use the wifi interface, no help.
Is there any debug flag that can be turned on so we can see anything happening, or write to a log file, anything at all so i can give more info about what's going on ?
Thanks.

FooDeas, 2019-05-26:
I'd check the following first in this order: Flashing activity LED? HDMI out? DHCP lease?

mocallins, 2019-05-26:
Nope, no flashing led activity.
I have a little usb expansion hub attached, and it has an led on it as well, and it flash once at power on, then again about a second later, then nothing.Don't worry i've tried it without this hub as well.Don't receive that rainbow screen either.
And i know its a working Pi, have connected via the usb port test with my computer, and I hear the ding, for new device.Also have used an 8GB card and all works fine with that.Just cannot get it to work with these 4GB cards i have.

FooDeas, 2019-05-26:
If the activity led doesn't show activity, it doesn't find the bootloader. Try to repartition the SD card properly.

mocallins, 2019-05-26:
Properly ?
I've used SD Formatter. Windows drive format options, and any number of other formating tools I've found / suggested.

FooDeas, 2019-05-26:
Then last idea:
Copy the content of the fat partition from your working 8GB card to the 4GB card and check if that shows any activity.

mocallins, 2019-05-26:
Sure, I'm game for anything, at this point, lol

mocallins, 2019-05-26:
I was also gonna tell you, that I've gone to the point of using a multifunction card reader, just to eliminate having to put the micro card in its larger adapter, no help, smh.

FooDeas, 2019-05-26:
You prooved that the cards are the problem...

mocallins, 2019-05-26:
Sorry I responded while I was in the process, just as an ennectdote.
Didn't work.And as a goof I put the 8GB card in there and it of course booted.

FooDeas, 2019-05-26:
I can't help you with these cards. Maybe you could check them with [Check Flash for Windows](http://mikelab.kiev.ua/index_en.php?page=PROGRAMS/chkflsh_en) or similar and face your seller with the probably bad results.

closed 2019-05-26 by FooDeas