FooDeas/raspberrypi-ua-netinst #171: Buster-support

davralin, 2019-06-25:
https://www.raspberrypi.org/blog/buster-the-new-version-of-raspbian/

Would simply specifying release=buster be enough?
I'd love to test, but my pi's are in hard-to-reach places, so I'd rather not pick them down if it fails.

kpfleming, 2019-06-25:
The first step in testing will be to dist-upgrade existing systems to the new release, and ensure that everything works properly. If your Pis are in 'hard to reach' places I'm not sure why you'd want to do a re-installation anyway...

davralin, 2019-06-26:
> If your Pis are in 'hard to reach' places I'm not sure why you'd want to do a re-installation anyway...
Because I prefer to stick with the latest stable, not oldstable, and my pi's are immutable.

Anyway, I foud an old Pi and did some testing.
if I simply change release too buster, I get this during the install.
`  I: Good signature from "Mike Thompson (Raspberry Pi Debian armhf ARMv6+VFP) <mpthompson@gmail.com>"
  P: Parsing Release
  P: Retrieving Packages.xz
  P: Validating Packages.xz
  P: Parsing Packages
  W: parser_rfc822: Iek! Don't find end of value!
  E: Internal error: download
`
Clean install with stretch, and then dist-upgrading works fine.

trvboro, 2019-06-26:
Same here: installation fails when release is set to "buster" with error "parser_rfc822: Iek! Don't find end of value!".

FooDeas, 2019-06-26:
It's a bug in cdebootstrap. I'll have to check if there is a fixed version available.

tdobes, 2019-06-26:
Upstream bugs:
- https://bugs.debian.org/904699
- https://bugs.debian.org/554444

It looks like cdebootstrap-static still needs to be rebuilt.

FooDeas, 2019-08-04:
I ported the installer to buster today and finally started the installation a few minutes ago.
If it succeeds, I'll post an RC tomorrow!

FooDeas, 2019-08-04:
Test ToDo list:
- installation via WiFi
- common compatibility with stretch, jessie, wheezy
- f2fs compatibility with stretch, jessie, wheezy

FooDeas, 2019-08-05:
RC1 is released!
Feel free to use it, have a look at the test ToDo list to support us and report the results.

ZwergNase, 2019-08-05:
Just tried WiFi install on my RPi3B with this options:
  preset=server
  firmware_packages=1
  release=buster
  ifname=wlan0
  wlan_country=DE
  wlan_ssid="xxxxxxxxxxxx"
  wlan_psk=xxxxxxxxxxxx

and received this error: 
wpa_supplicant: error while loading shared libraries: libnl-route-3.so.200: cannot open shared object file: no such file or directory

Hope this helps...

Cheers,
Oliver

FooDeas, 2019-08-05:
Fast report, fast bugfix: I deployed a new initramfs in the assets. Just replace the file and try again.
Thanks!

ZwergNase, 2019-08-05:
That did the trick- it's up and running! Thank you and good night, Oliver

FooDeas, 2019-08-11:
I do the last tests today and will release.

closed 2019-08-11 by FooDeas