FooDeas/raspberrypi-ua-netinst #172: RPi 4B support

mpmc, 2019-06-25:
Didn't see any info on support for the Pi 4 so thought I'd open an issue here for discussion. I have my Pi 4 here ready to roll for testing with ua-netinstaller :)

FooDeas, 2019-06-25:
Nice to hear! I'll provide you a test build this week. First of all, we should to check if the stretch kernel still works. Independently we should bring the installer to support buster. I hope that this will be not as hard as it was with stretch from jessie.

FooDeas, 2019-08-04:
It was a bit more work than thought but I think, I'll provide a test release tomorrow. With #171 the kernel of the 4B will be fully supported.
I'm counting on your feedback!

FooDeas, 2019-08-05:
RC1 is released!
Feel free to use it, have a look at the test ToDo list to support us and report the results.

mpmc, 2019-08-07:
Ok I've finally had time to test on my 4B 4GB. It works!

full log here: https://gist.github.com/mpmc/6ae1b9a01592f3374b11ac23afbe6a4e

FooDeas, 2019-08-07:
Could you do the installation with WiFi again and check if the connection is working afterwards, too?

mpmc, 2019-08-09:
> 
> 
> Could you do the installation with WiFi again and check if the connection is working afterwards, too?

Just saw this, I'll give it a go and get back to you :)

mpmc, 2019-08-10:
- [x]  Installation over wifi works - with the replacement raspberrypi-ua-netinst.cpio.gz
- [x]  Wifi AFTER install works!

```
mark@rpi4:~$ sudo ifconfig
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether dc:a6:32:03:8d:8b  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.153  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::dea6:32ff:fe03:8d8c  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:03:8d:8c  txqueuelen 1000  (Ethernet)
        RX packets 149  bytes 23816 (23.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 81  bytes 12875 (12.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

mark@rpi4:~$ uptime
 10:30:55 up 1 min,  1 user,  load average: 0.08, 0.03, 0.01
```

Awesome!

FooDeas, 2019-08-11:
Nice! Thanks a lot!

closed 2019-08-11 by FooDeas

dinosore, 2019-08-20:
Thanks for getting this sorted before my 4B arrives. :-)