FooDeas/raspberrypi-ua-netinst #175: Cannot connect to 802.11ac 5g wifi on RPi 4B

thomas201008, 2019-08-31:
I configured raspberrypi-ua-netinst V2.4.0 with **802.11ac 5g wifi**, instead of 2.4g or wired, then powered on my **RPi 4B**, but it stuck on "Configuring wlan0 with DHCP ...". 

On the same way, **RPi 3B+** is OK. And with Raspbian Buster Lite from raspberrypi.org, RPi 4B  is OK, too.

What's more, I reduced the OS with boot partition, static busybox, and static wpa_supplicant, nothing else. Again, RPi 3B+ can connect to 5g wifi, but RPi 4B cannot.

FooDeas, 2019-08-31:
I have no model 4, so I cannot check that. You could try to set the wifi country. Otherwise I'd suggest to build the installer by yourself, hoping that the repositories contain a newer kernel or packages that resolve the issue.
Please report!

thomas201008, 2019-09-02:
Of course, I configured country=US or country=CN.
And I used the same microSD with the same copy of binary programs and configuration, and in the same location, with kernel 4.19.57, 4.19.58 and 4.19.66 from https://github.com/raspberrypi/firmware. Resulting:
                                                 80211ac 5G                   80211n 2.4G
                                             RPi3B+     RPi4B             RPi3B+      RPi4B
Raspbian Buster                    OK            OK                  OK            OK
DietPi Buster                         OK            OK                  OK            OK 
raspberrypi-ua-netinst         OK            failed              OK             OK
my mininal linux                  OK            failed              OK             OK

[RPi3Bplus.wpa.log.txt](https://github.com/FooDeas/raspberrypi-ua-netinst/files/3564209/RPi3Bplus.wpa.log.txt)
[RPi4B.wpa.log.txt](https://github.com/FooDeas/raspberrypi-ua-netinst/files/3564210/RPi4B.wpa.log.txt)

**Please refer to the wpa_supplicant log or RPi3B+ and RPi4B.**

**log on RPi3B+, scanned 5g wifi and found 5g wifi**
......
nl80211: Scan included frequencies: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 **5745 5765 5785 5805 5825**
wlan0: Event SCAN_RESULTS (3) received
wlan0: Scan completed in 2.805478 seconds
nl80211: Received scan results (6 BSSes)
wlan0: BSS: Start scan result update 1
**wlan0: BSS: Add new id 0 BSSID e8:fc:af:98:3e:6f SSID 'DEBIAN8' freq 5745**
wlan0: BSS: Add new id 1 BSSID c8:3a:35:03:44:58 SSID 'Tend_808' freq 2422
wlan0: BSS: Add new id 2 BSSID e8:fc:af:98:3e:6b SSID 'Debian8' freq 2417
wlan0: BSS: Add new id 3 BSSID 8c:a6:df:6d:26:87 SSID 'TP-LINK_2687' freq 2412
wlan0: BSS: Add new id 4 BSSID 68:89:75:08:29:78 SSID '' freq 2412
wlan0: BSS: Add new id 5 BSSID 40:16:7e:2c:c9:c0 SSID 'DebianEX' freq 2447
BSS: last_scan_res_used=6/32
wlan0: New scan results available (own=1 ext=0)
......

**log on RPi4B, scanned 5g wifi but found 2.4g only**
......
nl80211: Scan included frequencies: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484 5170 5180 5190 5200 5210 5220 5230 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 **5745 5765 5785 5805 5825**
wlan0: Event SCAN_RESULTS (3) received
wlan0: Scan completed in 2.559010 seconds
nl80211: Received scan results (6 BSSes)
wlan0: BSS: Start scan result update 2
BSS: last_scan_res_used=6/32
wlan0: New scan results available (own=1 ext=0)
wlan0: Radio work 'scan'@0x1d377e8 done in 2.582936 seconds
wlan0: radio_work_free('scan'@0x1d377e8): num_active_works --> 0
wlan0: Selecting BSS from priority group 0
wlan0: 0: c8:3a:35:03:44:58 ssid='Tend_808' wpa_ie_len=24 rsn_ie_len=20 caps=0x411 level=-60 freq=2422 
wlan0:    skip - SSID mismatch
wlan0: 1: e8:fc:af:98:3e:6b ssid='Debian8' wpa_ie_len=22 rsn_ie_len=20 caps=0x1411 level=-39 freq=2422 
wlan0:    skip - SSID mismatch
wlan0: 2: 8c:a6:df:6d:26:87 ssid='TP-LINK_2687' wpa_ie_len=22 rsn_ie_len=20 caps=0x431 level=-71 freq=2412 
wlan0:    skip - SSID mismatch
wlan0: 3: 24:05:0f:79:29:c8 ssid='360\xe5\x85\x8d\xe8\xb4\xb9WiFi-C8' wpa_ie_len=24 rsn_ie_len=20 caps=0x431 level=-72 freq=2437 
wlan0:    skip - SSID mismatch
wlan0: 4: 24:05:0f:a6:f8:2e ssid='160WiFi_F82F' wpa_ie_len=0 rsn_ie_len=20 caps=0x431 level=-76 freq=2412 
wlan0:    skip - SSID mismatch
wlan0: 5: 40:16:7e:2c:c9:c0 ssid='DebianEX' wpa_ie_len=22 rsn_ie_len=20 caps=0x411 level=-79 freq=2427 
wlan0:    skip - SSID mismatch
wlan0: No suitable network found
......

is it possible, RBi4B requires other kernel modules, not only brcmfmac, or some kernel parameters needs to be tuned, or some else, for example, DBus.

Thanks

FooDeas, 2019-09-02:
crda could also be a keyword...
Is 5ghz wifi working after you installed it with cable or 2.4g?

thomas201008, 2019-09-09:
Yesterday, I had a try with another 5g router at **freq 5180MHz, and it worked.**

nl80211: Scan included frequencies: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484 5170 5180 5190 5200 5210 5220 5230 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825
wlan0: Event SCAN_RESULTS (3) received
wlan0: Scan completed in 2.550793 seconds
nl80211: Received scan results (10 BSSes)
wlan0: BSS: Start scan result update 1
**wlan0: BSS: Add new id 0 BSSID 54:df:24:22:7f:78 SSID 'CMCC-D995-5G' freq 5180**
wlan0: BSS: Add new id 1 BSSID 02:4b:f3:0a:8b:48 SSID 'Xiaomi_A50F' freq 2462
wlan0: BSS: Add new id 2 BSSID 54:df:24:22:7f:77 SSID 'CMCC-D995' freq 2472
......

[wpa.5180MHz.log.txt](https://github.com/FooDeas/raspberrypi-ua-netinst/files/3588334/wpa.5180MHz.log.txt)

Previously, the 5g wifi signal I tested with was at freq 5745MHz, but it is a production system, I could not modify it except for emergency. It's a ASUS router. I found if I configured the region of the router as CN, the bands will limited from 149 to 165, and US will be more, from 36 to 48, 149 to 165.
[image]
[image]

That's to say, 
with raspberrypi-ua-netinst v2.4.0, RPi3B+ can connect to 5745MHz and 5180MHz, and might be more bands. But RPi4B cannot connect to 5745MHz. On the other hand, with the Official OS Respbian Buster, RPi4B can connect to 5745MHz.

Maybe, just maybe, with raspberrypi-ua-netinst v2.4.0, RPi4B will not work at the bands from 
 freq 5490MHz. But if a router will select the bands automatically at any time, the connection may go wrong, and that would be a problem. So far, I haven't found why Respbian Buster works.

Thanks,

FooDeas, 2019-09-09:
There are some weird things:
1. There is nothing special (DFS or something else) with channel 149 so there should be no problem. I can't debug this but I agree that there may be missing packages.
2. In China, your router should provide many more than the five 5Ghz channels. (13x20MHz or 6x40MHz or 1x80MHz)

thomas201008, 2021-01-28:
1. The allowed 5GHz WiFi channels in China are 149, 153, 157, 161 and 165, according to the following link.
https://www.arubanetworks.com/vrd/OutdoorMIMOVRD/wwhelp/wwhimpl/common/html/wwhelp.htm#context=OutdoorMIMOVRD&file=AppA.html

2. To solve this issue in RPi4B, /lib/firmware/regulatory.db and regulatory.db.p7s are required.

closed 2023-08-19 by FooDeas