FooDeas/raspberrypi-ua-netinst #180: Install hangs on package install

yatzy, 2019-09-22:
Hi

For some reason the installation hangs when large number of packages are listed in installer-config.txt. I'm trying to install these on Raspberry Zero W. 

packages in my installer-config.txt:
packages=python3,sudo,curl,python3-pip,ca-certificates,systemd,libjasper-dev,libqtgui4,python3-dev,python3-pyqt5,libilmbase-dev,htop,git,nano,fail2ban,libhdf5-dev,libatlas-base-dev,libavcodec57,libavformat-dev,libgstreamer1.0-dev,libsm6,libxext6,libwebp6,libswscale-dev,libopenexr-dev

The hang happens while  selecting and unpacking the packages. What could be the reason for this?

FooDeas, 2019-09-22:
In most cases, the raspbian/raspberrypi repository causes hanging. I'll give it a try on my Zero W later to check the installer with your package list.
It may be helpful to have your anonymised installer-config.txt.

yatzy, 2019-09-22:
Here's the anonymized config I'm using 
[installer-config.txt](https://github.com/FooDeas/raspberrypi-ua-netinst/files/3639827/installer-config.txt)

This is the phase the installer hangs, sorry for bad picture: https://photos.app.goo.gl/nKWg8ifi4rGufoUeA

FooDeas, 2019-09-22:
I tried it and found the issue in the log: `E: Package 'libavcodec57' has no installation candidate`
If you want to go the easy way, choose the previous installer version v2.3.0 which uses stretch by default.

yatzy, 2019-09-22:
Oh I'm sorry I didn't mention I'm already using the 2.3.0 because of the aforementioned reason.

FooDeas, 2019-09-22:
Thanks for the late info! :)
I'll check that again.

yatzy, 2019-09-28:
Hi, just wanted to ask whether we have any new information about this issue?

FooDeas, 2019-10-02:
I have to check it this weekend. I need to fix the WiFi adapter of the Zero with buster installer, too.

closed 2023-08-15 by FooDeas