FooDeas/raspberrypi-ua-netinst #181: could not download deb 10 u1 armhf.deb

uplamartya, 2019-10-01:
[image]

Kindly provide any solution. using 3B+

Mausy5043, 2019-10-02:
Could be congestion at the server. Have you tried it again at a later time?

closed 2019-10-02 by FooDeas

uplamartya, 2019-10-02:
> Could be congestion at the server. Have you tried it again at a later time?
> 
> "If at first you don't succeed..." ;-)

Multiple times . Every time i got the same error

FooDeas, 2019-10-02:
It is a connection or server problem. Set installer_pkg_downloadretries to a high value that allows more retries instead of 5.

uplamartya, 2019-10-02:
@FooDeas , ok thanks. Will let you know after trying this .