FooDeas/raspberrypi-ua-netinst #184: Pre-add .deb files in the image

PUsmanKhan, 2020-01-07:
I opened the update.sh file and saw multiple package being adding up as packages+=("xxxx"). I want to add heavy packages that take a lot of time to download inside the image while running the installer. This way I can remove those packages from installer-config.txt. I guess this would increase a bit of size of the image but I don't want to wait an hour for the image to set up.

I added the respective packages by adding packages+=("chromium-browser") etc etc. I can see them being downloaded when build.sh is run but when the image is fully set up I cannot see chromium or any other packages installed in the Pi. Can you please help me. I guess I am doing something wrong or this script has a bug?

I can alternatively use custom-files, put them in a directory and use post-install script to install them but I think install.sh would be a great option to install them.

Kindly look into this issue. You have solved a lot of problems by creating this repo.

A big Fan of your work
Thanks

kpfleming, 2020-03-16:
update.sh downloads packages which will be used to build the installer image itself; if the build.sh script doesn't unpack the downloaded image and copy parts of it into the image, then downloading the package won't have any effect, as you noticed.

This change will require modifying build.sh to support a list of packages to be copied into a directory in the installer image, and then modifying install.sh to drop those packages into the proper location in the installed system (after cdebootstrap has been run). Finally, a post-install script could request those packages to be installed, and since they are already present in the filesystem no download would be required.

Keep in mind that the installed system uses the latest versions of packages which are requested; if the installer image contains an older version of a package, it wouldn't be used. The only way around that would be to explicitly install each .deb file, which custom-files+post-install could be used to do (and you have to be careful that you've obtained all the dependencies for those packages as well).

It wouldn't be hard to write a script which does all of this for you: downloads the package, puts it into the custom-files directory, and adds a post-install script for it.

Scientist786, 2020-03-16:
I can do that by custom-files and post-install script but wouldn't it be better if we had it done in build.sh like you said 

> This change will require modifying build.sh to support a list of packages to be copied into a directory in the installer image

I think this is an acceptable and easy way because if we are copying .deb files through custom-files, we might not have the latest version of the packages while build.sh would always download the latest version just like update.sh. I am a bit new in bash scripting so I really don't know how I would accomplish that.

Right now I have used the custom-files and post-install script. If you want, I can share the code with you but you already know how I did it.

FooDeas, 2023-08-15:
The "Bring your own files" in combination with the "Post installer script" is the designated way. Another possibility to speed things like this up if you need the installer more often is to use an apt cacher.

closed 2023-08-15 by FooDeas