FooDeas/raspberrypi-ua-netinst #191: Install files feature.

mocallins, 2020-04-11:
Is the installer supposed to copy the files from  ..\config\files\root\accordng-to-install_files.list ?

I've looked in multiple places, they only seem to reside in /boot/* on the trget system.

Thanks

kpfleming, 2020-04-11:
The `.list` file goes into config/files; the files to be copied go into config/file/root (and further subdirectories based on the locations they should be copied into). You have to tell the installer which files you want copied, and their owner/group and permissions, which you do by creating the `.list` file.

mocallins, 2020-04-12:
Yep, i think i did all that, files don't get copied.

FooDeas, 2020-04-12:
Attach/post your "accordng-to-install_files.list" and post the exact location of one of your files.

mocallins, 2020-04-13:
[image]

[image]

[image]

kpfleming, 2020-04-13:
Do you have your editor (ConTEXT) configured to produce Unix line endings instead of Windows line endings? If not that could be a problem.

Also, after the installation is complete you can look at the log file in the /boot directory to see what happened during the installation, you should be able to find the section towards the end where the install_files.list file was opened and processed.

mocallins, 2020-04-13:
I just checked and changed that file, in my editor, to use unix (LF), it was  dos (crlf)  :~
can i just run again as a reinstall ? or have to  redo the whole thing ?

mocallins, 2020-04-13:
Also, does netinst contain all the other firmware-netcard-package, so the it can recognize any supported adapter, or is that possible ?

Any future enhancements to change the shell, i like to use tcsh ?

FooDeas, 2020-04-13:
[image]
Mystery solved?

mocallins, 2020-04-14:
AAaaarrrrgggghhhh, yep you're right, gaaahhh, sorry about that.

closed 2020-04-14 by mocallins