FooDeas/raspberrypi-ua-netinst #192: Kernel modules for the kernel version "4.19.97-v7+" could not be found.

unclehowell, 2020-04-21:
UA-NETINST Works fine on the Raspberry Pi4, but I want to create an image which will work on both the RPI3 and RPI4. However, no matter what I try, I keep getting this error on the Raspberry Pi 3 (B). Even when I scrub all the Raspberry Pi config and just focus on trying to get this to work on the Raspberry Pi B, I keep arriving at this error. The killer is, I have to go through the 20 minute process every time just to see this error. What's the solution please ? 

Kernel modules for the kernel version "4.19.97-v7+" could not be found. Searching for alternatives...
  Using modules of kernel version "4.19.58-v7+".
modprobe: can't load module rfkill (kernel/net/rfkill/rfkill.ko): invalid module format
Waiting for wlan0... 1.. 2.. 3.. 4.. 5.. 6.. 7.. 8.. 9.. 10.. 11.. 12.. 13.. 14.. 15.. 16.. 17.. 18.. 19.. 20.. 21.. 22.. 23.. 24.. 25.. 26.. 27.. 28>

Error: The installation could not be completed!

FooDeas, 2020-04-22:
Please use the current release with the updated kernels and report.

unclehowell, 2020-04-22:
The one just posted an hour ago. Will do. Thanks

closed 2020-04-23 by unclehowell