FooDeas/raspberrypi-ua-netinst #195: Add support for 64-bit

davralin, 2020-06-04:
https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=275370

Would be swelll if I could use ua-netinst to bootstrap 64-bit-images.

I realize that this will probably take a while though...

kpfleming, 2021-01-01:
I'm working on this right now (doing some final testing before posting an issue for discussion, since the changes are extensive). One thing to note: I'm doing this by adding support for Debian as a base (both for the installer and the installed system), I'm not doing any work to try to install 64-bit Raspbian or Raspberry Pi OS (as I prefer to just use plain Debian).

Rocco83, 2021-04-08:
Any news? Or even, how to test it out?

ovimano, 2021-07-04:
Hello!
Can I test this pull request?

davralin, 2022-06-05:
For antoine still looking for something, this might be a viable solution:

https://raspi.debian.net/

closed 2022-06-05 by davralin