FooDeas/raspberrypi-ua-netinst #198: Installer doesn't like `DIR=/var/run/wpa_supplicant GROUP=netdev` in wpa_supplicant.conf

Mausy5043, 2020-07-11:
Here is the log while trying to build on a RPi 3B+ via WiFi:

```
==================================================
raspberrypi-ua-netinst
==================================================
Revision 685148f
Built on Sat 11 Jul 2020 09:55:22 AM CEST
Running on Raspberry Pi version 3 Model B+
==================================================
https://github.com/FooDeas/raspberrypi-ua-netinst/
==================================================
Remounting TempFS... OK
Mounting boot partition... OK
Copying boot files... OK
Executing installer-config.txt...
OK

Searching for deprecated variables...

Checking if config.txt needs to be modified before starting installation...
  Setting HDMI options... OK
OK

Unmounting boot partition... OK

Network configuration:
  ifname = wlan0
  ip_addr = dhcp
  online_config = 

Waiting for wlan0... 1.. OK
Starting wpa_supplicant...
  Successfully initialized wpa_supplicant
  CTRL: Invalid group 'netdev'
  Failed to initialize control interface 'DIR=/var/run/wpa_supplicant GROUP=netdev'.
  You may have another wpa_supplicant process already running or the file was
  left by an unclean termination of wpa_supplicant in which case you will need
  to manually remove this file before starting wpa_supplicant again.
  
  nl80211: Failed to open /proc/sys/net/ipv6/conf/wlan0/drop_unicast_in_l2_multicast: No such file or directory
  nl80211: Failed to set IPv6 unicast in multicast filter
  nl80211: Failed to open /proc/sys/net/ipv6/conf/wlan0/drop_unicast_in_l2_multicast: No such file or directory
  nl80211: Failed to set IPv6 unicast in multicast filter
  nl80211: deinit ifname=wlan0 disabled_11b_rates=0
  nl80211 driver didn't work. Trying generic driver (wext)...
  Successfully initialized wpa_supplicant
  ioctl[SIOCSIWENCODEEXT]: Invalid argument
  ioctl[SIOCSIWENCODEEXT]: Invalid argument
  CTRL: Invalid group 'netdev'
  Failed to initialize control interface 'DIR=/var/run/wpa_supplicant GROUP=netdev'.
  You may have another wpa_supplicant process already running or the file was
  left by an unclean termination of wpa_supplicant in which case you will need
  to manually remove this file before starting wpa_supplicant again.
  

Error: The installation could not be completed!
```

 * Removing the line `DIR=/var/run/wpa_supplicant GROUP=netdev` from the `wpa_supplicant.conf` results in a successful build.
 * Interestingly, re-adding the line on the installed system works perfectly.

yatzy, 2020-12-07:
Facing same issue

Mausy5043, 2021-07-10:
The issue @ https://github.com/guysoft/OctoPi/issues/636 seems to suggest that creating a group 'netdev' might fix this.

Mausy5043, 2021-07-11:
I've created a pull request that fixes this issue. @FooDeas please consider this for merging.

PR #222 (2021-07-11): create netdev group
This should fix #198.
closed 2021-07-11 by FooDeas