FooDeas/raspberrypi-ua-netinst #199: Installer not booting anymore

Mausy5043, 2020-07-25:
Recent builds (as far as I can tell since the past few days) do not produce a bootable builder anymore.
The only error I see during the build is this one:

```
cp: cannot stat 'tmp/lib/modules/5.4.51-v7+/kernel/drivers/hwmon/hwmon.ko': No such file or directory
```

Any ideas where I might need to start looking?

FooDeas, 2020-07-25:
Seems to be a changed package content.
Update and build hast to be checked. I can do that next week.

kpfleming, 2020-07-25:
Yes, there was a major kernel change (from 4.19 to 5.4) this week.

Mausy5043, 2020-07-25:
OK thanks.
To provide you with a bit more info, although I think it wont add a lot, I've managed to get one of the Pi's hooked up to a screen and get a screenshot of the attempted boot.

https://photos.app.goo.gl/95FrM14AZQ7cymL59

Mausy5043, 2020-07-25:
> Seems to be a changed package content.
> Update and build hast to be checked. I can do that next week.

Thanks. would appreciate that very much.

FooDeas, 2020-07-26:
It is time to develop an automatic resolution of package dependencies and file/link handling.
I hope this will not add too much unnecessary content...

Mausy5043, 2020-08-08:
I tried building using ```master``` but that doesn't work either.

Mausy5043, 2020-08-08:
A new build is still running, but I'm confident that patch #200 works.
Please review & commit.

FooDeas, 2020-08-08:
Thank you! I'm still at scripting the dependency solver. It's horrible...

PR #200 (2020-08-08): fix for missing `hwmon.ko`
Fixes #199 .
closed 2020-08-08 by FooDeas