FooDeas/raspberrypi-ua-netinst #203: Fastest possible boot time?

lukehutch, 2020-09-15:
I was wondering what work has been done to optimize boot time with this Raspbian distribution. I was going to follow the tips at the following link, then I discovered this fork of raspberrypi-ua-netinst, and decided to try one of the small installs. Can any of the following tips be incorporated to create an ultra-fast-booting distribution?

https://himeshp.blogspot.com/2018/08/fast-boot-with-raspberry-pi.html

kpfleming, 2021-01-01:
This is an installer, not a distribution, so it doesn't really control boot time of the installed system. It's certainly possible that 'quick boot' optimizations could be included in the installer script so that the resulting system boots more quickly, but they'll have to be opt-in features if they change the behavior of the system since this installer is supposed provide a system which is very similar to the base Raspbian installation (in minimal form, of course).

lukehutch, 2021-01-01:
Fair enough, though I assumed that just as there are small installations that you offer, there could be even smaller installations. If you're not interested in any of the optimizations I linked, or this installer is not the place for them, feel free to close. Thanks!

kpfleming, 2021-01-02:
I'm not the keeper of this project, just a happy user who has contributed a bunch of stuff, so my opinion is only mine :-)

If you configure the installer to build a 'minimal' system you're going to get the smallest possible Debian-derived system.

Looking at the link you posted (it's based on starting with a full Raspbian system, which is not what this installer provides):

* Section 1: most of that is probably relevant, but many of those settings are very specific to the hardware being used (for example boot_delay=0 doesn't work with SD cards or USB drives that are slow to enumerate in the kernel). It's already possible to specify additional Device Tree overlays to be applied in the configuration, so if there's a desire to disable the WiFi hardware to decrease boot times that's available, but if it was commonly desired an 'advanced' configuration option for it would be a good idea.

* Section 2: This is already an option.

* Section 3: These steps disable the network, SSH server, NTP client (most of the rest of the services listed in this section are not installed by this installer). Disabling those doesn't make sense for a (probably headless) networked system, which is the most common use of this installer.

* Section 4: Not relevant.

* Sections 5-7: This is again very system specific; I use a self-compiled and tuned kernel on my Pi systems (after the initial installation), but it's not the right choice for everyone, and definitely not something an installer can offer.

* Section 8: This will break kernel and firmware updates, and also make it more difficult to test configuration options (as in section 1).

* Section 9: Not relevant.

Overall, starting with this installer is going to give you the best chance to build a minimal system which boots as quickly as any normal Debian-based system can boot; going beyond that point is going to require tailoring the configuration to the specific application you have in mind. If there are specific suggestions you have and can provide evidence that those changes make a measurable difference in startup times, I'm sure everyone here would be happy to discuss them.

lukehutch, 2021-01-02:
OK, that's fair, thanks for taking a look and going over the post section by section. I'll close this, since I really don't know enough about Raspbian to suggest anything else. I guess it's too much to hope for Arduino-like startup times with a Raspberry Pi, at least running Raspbian or similar! (The current 30 second-ish startup time is a showstopper for an application I'm trying to use RPi for...)

closed 2021-01-02 by lukehutch

kpfleming, 2021-01-02:
Yes, that's true, there's a huge difference between any Linux-based system and an Arduino. However, if you're using an RPi for a dedicated (single) application which doesn't require a complex system to support it, you could consider using the busybox-based system (which is constructed by `build.sh`) as a starting point. If you provide a custom 'install' script which doesn't actually do any installation, but instead starts and runs your application, you'd have a very lightweight system with a very fast startup time (and as a bonus, it would run entirely out of RAM, only using the SD card for boot).

lukehutch, 2021-01-02:
@kpfleming that actually sounds super useful, and I think many people would benefit from something like that. I don't know Raspbian well enough to put something like this together. Would you be able to write up a quick tutorial about how to build this, please?

kpfleming, 2021-01-02:
No, that's a many hour project, and honestly, what you're asking for isn't Raspbian (or Debian, or any other major distribution).

You might find it helpful to read this [blog](https://medium.com/@hungryspider/building-custom-linux-for-raspberry-pi-using-buildroot-f81efc7aa817) and do some additional Internet searches for 'buildroot Raspberry Pi'. 'buildroot' is a system intended exactly for what you want: building a stripped-down Linux installation for a dedicated application.

kpfleming, 2021-01-02:
Yocto is another project in that area, here's a [post](https://jumpnowtek.com/rpi/Raspberry-Pi-Systems-with-Yocto.html) about that one.

lukehutch, 2021-01-03:
@kpfleming Awesome, thanks for the pointers!