FooDeas/raspberrypi-ua-netinst #204: Is this distribution working on RPI 1b?

positron96, 2020-09-18:
I've tried several times to reflash an SD card with this distribution, but was always getting LED blinking 4 times (something related to `start.elf` broken or missing) on my old RPI 1b (from something like 2012). I solved it by replacing start.elf with one from `raspbian-ua-netinst` project, latest version of Dec 2019 (or maybe some other dark magic solved it).

I am posting this to hint that there might be a problem with boot files on oldest RPIs. Or maybe my SD card is wearing out, though I've formatted it several times and did a clean copy from both zip file and img.gz file.

FooDeas, 2021-01-04:
Should be considered in upcoming release.

FooDeas, 2023-08-15:
Successfully tested with v2.5.0 on a model 1B+ today so a model 1B will also work.

closed 2023-08-15 by FooDeas