FooDeas/raspberrypi-ua-netinst #205: Doesn't boot reliably on RPi 4B

lukehutch, 2020-09-24:
I tried about 20 times to get raspberrypi-ua-netinst to boot on a Raspberry Pi 4B. I succeeded only once. Normally the screen is black (no HDMI signal) and the activity light blinks a few times then turns off. Booting the official Raspberry Pi OS has no problems on the same hardware.

kpfleming, 2021-01-01:
Did you build your own installer image, or use a pre-built image?

lukehutch, 2021-01-01:
It was a pre-built image, whatever was available at the time I posted this bug report.

kpfleming, 2021-01-02:
Yeah, unfortunately the installer images get outdated rather quickly, as they include a copy of the kernel which was current at that time.

If you've got the time and energy, building your own installer from a fresh copy of the 'devel' branch will probably produce a better result.

lukehutch, 2021-01-02:
OK -- I already moved on to regular Raspbian, shipped the product to testers, and moved on to other projects, so I'll close this -- I just filed in the hopes that it helped locate an issue. Thanks!

closed 2021-01-02 by lukehutch