FooDeas/raspberrypi-ua-netinst #214: Booting release 2.4.0+ (caf7423) on RPi 3B+ using WiFi fails

leilabd, 2021-03-04:
Booting release 2.4.0+ (caf7423) on RPi 3B+ using WiFi fails.
From the error log:

```
Network configuration:
  ifname = wlan0
  ip_addr = dhcp
  online_config = 

Kernel modules for the kernel version "5.10.11-v7+" could not be found. Searching for alternatives...
  Using modules of kernel version "4.19.97-v7+".
modprobe: can't load module rfkill (kernel/net/rfkill/rfkill.ko): invalid module format
modprobe: can't load module brcmfmac (kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko): invalid module format
Waiting for wlan0... 1.. 2.. 3.. 4.. 5.. 6.. 7.. 8.. 9.. 10.. 11.. 12.. 13.. 14.. FAILED

Error: The installation could not be completed!
```

On looking through closed issues, I saw that the 3B+ WiFi had been a problem with version 2.2.0 and was then fixed, so I tried some previous versions: the 3B+ Wifi worked on v2.2.1 and v2.3.0 but not on v2.4.0 or v2.4.0+.

FooDeas, 2023-08-15:
Please check v2.5.0.

closed 2023-08-15 by FooDeas

leilabd, 2023-08-21:
I can confirm that v2.5.0 boots fine on my RPi 3B+ with WiFi (ethernet not currently connected). I shall be very pleased to have my print server running a current release of the operating system.

Many thanks.