FooDeas/raspberrypi-ua-netinst #216: Annoying Errors During Build

unclehowell, 2021-03-08:
Three Annoying Errors During Build: Can something be done to solve these ? 

a. WARNING: apt does not have a stable CLI interface. Use with caution in scripts
b.  /usr/bin/apt-key: 57: /usr/bin/apt-key: cannot create /dev/null: Permission denied
c. W: Failed to fetch http://archive.raspberrypi.org/debian/dists/buster/InRelease  Temporary failure resolving 'archive.raspberrypi.org'

Scientist786, 2021-03-24:
I am having the same issues. I solved hwmon error but the rest are still there.

Mausy5043, 2021-04-04:
This worked for me: ```sed -ibak s/\/hwmon\.ko/\/raspberrypi-hwmon\.ko/ build.sh```

Mausy5043, 2021-04-11:
Hmm 😞 .   
Now also running into `cp: cannot stat 'tmp/lib/modules/5.10.17-v8+/kernel/drivers/net/phy': No such file or directory`

Scientist786, 2021-04-11:
@Mausy5043 I already told you to comment out those lines. Read my previous comment. I am using the generated image for a month, no issue faced till now.

Mausy5043, 2021-04-11:
@Scientist786 : I understand that your fix works for **you**. But for the general case this is not enough.

The installer will hold four `net` directories in `raspberrypi-ua-netinst/build_dir/tmp/lib/modules`:
```
$ find . -wholename "*/kernel/drivers/net"
./5.10.17-v8+/kernel/drivers/net
./5.10.17-v7+/kernel/drivers/net
./5.10.17-v7l+/kernel/drivers/net
./5.10.17+/kernel/drivers/net
```
and only one of them holds a `phy` directory:
```
$ find . -wholename "*/kernel/drivers/net/phy"
./5.10.17+/kernel/drivers/net/phy
```
So, although your fix may work, it will fail for those using the `5.10.17+` kernel.

Mausy5043, 2021-04-11:
This line needs to be changed: 
https://github.com/FooDeas/raspberrypi-ua-netinst/blob/15b8f30d8b436f081b0d49cf55ca2af84b0a112c/build.sh#L41

to:

```
eval cp --preserve=xattr,timestamps -r "${1//kernel\*/${kernel}}" "${2//kernel\*/${kernel}}"  || true
```

This will still output the copying error during the build, but it will not abort it.

And to also tackle OPs final error. This line:
https://github.com/FooDeas/raspberrypi-ua-netinst/blob/15b8f30d8b436f081b0d49cf55ca2af84b0a112c/build.sh#L221

should be changes to `cp --preserve=xattr,timestamps tmp/sbin/mkfs.btrfs rootfs/bin/` because the executable was moved.

commit 4bf6df07 (2021-04-11): Merge pull request #217 from Mausy5043/fix216

This should fix #216
closed 2021-04-11 by FooDeas